ros2/rosbag2 #1183: Unable to set QoS profile from node in ROS2 Humble

laurafbec, 2022-11-27:
## Description
I'm trying to set QoS policies to the /map topic from a node by using:
`topic_info_map = rosbag2_py._storage.TopicMetadata(name='map', type='nav_msgs/msg/OccupancyGrid', serialization_format='cdr', offered_qos_profiles="- history: keep_all\n  depth: 5\n  reliability: reliable\n  durability: transient_local\n  deadline:\n    sec: 0\n    nsec: 0\n  lifespan:\n    sec: 0\n    nsec: 0\n  liveliness: system_default\n  liveliness_lease_duration:\n    sec: 0\n    nsec: 0\n  avoid_ros_namespace_conventions: false")
       `

or 
topic_info_map = rosbag2_py._storage.TopicMetadata(name='map', type='nav_msgs/msg/OccupancyGrid', serialization_format='cdr', offered_qos_profiles='qos_profile.yaml')

where the file qos_profile.yaml contains:

`map:
  history: keep_all
  depth: 5
  reliability: reliable
  durability: transient_local
  deadline:
    sec: 0
    nsec: 0
  lifespan:
    sec: 0
    nsec: 0
  liveliness: system_default
  liveliness_lease_duration:
    sec: 0
    nsec: 0
  avoid_ros_namespace_conventions: false`
       

### Expected Behavior
** clear and concise description of what you _expected_ to happen **

### Actual Behavior
When running `ros2 topic info /map --verbose`

I get these value for my node, being the durability equal to VOLATILE instead of TRANSIENT_LOCAL
`Node name: simple_bag_recorder

Node namespace: /

Topic type: nav_msgs/msg/OccupancyGrid

Endpoint type: SUBSCRIPTION

GID: 01.0f.65.91.32.6c.ce.54.01.00.00.00.00.00.13.04.00.00.00.00.00.00.00.00

QoS profile:

  Reliability: RELIABLE

  History (Depth): UNKNOWN

  Durability: VOLATILE

  Lifespan: Infinite

  Deadline: Infinite

  Liveliness: AUTOMATIC

  Liveliness lease duration: Infinite

`

Thanks in advance
## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. Logs print error '...' **

## System (please complete the following information)
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2022-12-01:
@laurafbec Could you please provide more details with exact steps about how to reproduce this issue or provide minimum valuable reproducer.

laurafbec, 2022-12-01:
Thanks @MichaelOrlov!

This is a minimal node to save /map messages.
I don't know the proper way to provide offered_qos_profiles to the Topic Metadata.
Thanks again

`import rclpy
from rclpy.node import Node
from rclpy.serialization import serialize_message
from nav_msgs.msg import OccupancyGrid

from os.path import join
from ament_index_python.packages import get_package_share_directory

from rclpy.qos import QoSProfile
from rclpy.qos import QoSDurabilityPolicy, QoSHistoryPolicy, QoSReliabilityPolicy

import rosbag2_py

class SimpleBagRecorderMap(Node):
    def __init__(self):
        super().__init__('simple_bag_recorder_map')

        pkg_bag_recorder_nodes_py = get_package_share_directory("bag_recorder_nodes_py")
        default_qos_dir = join(pkg_bag_recorder_nodes_py, "config", "qos_profile.yaml")
        print(default_qos_dir)

        # Writer object
        self.writer = rosbag2_py.SequentialWriter()

        # URI of the bag to create and the format
        storage_options = rosbag2_py._storage.StorageOptions(uri='my_bag', storage_id='sqlite3')

        # Default conversion options to store the message in the serialization format the are received in
        converter_options = rosbag2_py._storage.ConverterOptions('', '')
        self.writer.open(storage_options, converter_options)

         # QoS profiles definition
        map_qos = QoSProfile(
          durability=QoSDurabilityPolicy.TRANSIENT_LOCAL,
          reliability=QoSReliabilityPolicy.RELIABLE,
          history=QoSHistoryPolicy.KEEP_ALL,
          depth=5)

       

        # Topics' specification and writer's registering process
        topic_info_map = rosbag2_py._storage.TopicMetadata(name='map', 
                                                          type='nav_msgs/msg/OccupancyGrid', 
                                                          serialization_format='cdr', 
                                                          offered_qos_profiles=' history: keep_all\n  depth: 5\n  reliability: reliable\n  durability: transient_local\n  deadline:\n    sec: 0\n    nsec: 0\n  lifespan:\n    sec: 0\n    nsec: 0\n  liveliness: system_default\n  liveliness_lease_duration:\n    sec: 0\n    nsec: 0\n  avoid_ros_namespace_conventions: false')
        
       
        self.writer.create_topic(topic_info_map)

        # Subscriptions definition
        self.subscription = self.create_subscription(OccupancyGrid, 'map', self.topic_callback_map, map_qos)
        self.subscription

    # Callbacks functions to pass the serialized message to the writer
    def topic_callback_map(self, msg):
        self.writer.write('map', serialize_message(msg), self.get_clock().now().nanoseconds)

def main(args=None):
    rclpy.init(args=args)
    sbr = SimpleBagRecorderMap()
    rclpy.spin(sbr)
    rclpy.shutdown()

if __name__ == '__main__':
    main()`

MichaelOrlov, 2022-12-07:
Hi @laurafbec, I've tried to make analysis of your minimal reproducer and expectation from issue description.
If I am understand correctly the problem doesn't really relates to the rosbag2.
You are creating subscription from python with your own QoS profile and `ros2 topic info /map --verbose` shows that your QoS settings was not applied to the subscription. Right?

After some code review of your reproducer I think that problem is in second initialization of your subscription with default parameters. `self.subscription`
```
# Subscriptions definition
        self.subscription = self.create_subscription(OccupancyGrid, 'map', self.topic_callback_map, map_qos)
        self.subscription
```
Please note that we have some unit tests for `node.create_subscription(..)` with different QoS profiles.
https://github.com/ros2/ros2cli/blob/1cf39d4bfa33e11050ffb05b77c39c36cf071d6d/ros2topic/test/qos_subscription.py#L40-L82
Which is passing on CI (I hope :smile: )

laurafbec, 2022-12-07:
Thanks @MichaelOrlov!
My present problem is related to the QoS from the node's recorded topics when running `ros2 bag play`.
I've tried to define the profile by using the yaml file path or even the whole content but I don't know what is the proper way.

```
 # Topics' specification and writer's registering process
        topic_info_map = rosbag2_py._storage.TopicMetadata(name='map', 
                                                          type='nav_msgs/msg/OccupancyGrid', 
                                                          serialization_format='cdr', 
                                                          offered_qos_profiles=' history: keep_all\n  depth: 5\n  reliability: reliable\n  durability: transient_local\n  deadline:\n    sec: 0\n    nsec: 0\n  lifespan:\n    sec: 0\n    nsec: 0\n  liveliness: system_default\n  liveliness_lease_duration:\n    sec: 0\n    nsec: 0\n  avoid_ros_namespace_conventions: false')

```
Thanks again!

MichaelOrlov, 2022-12-07:
@laurafbec To be able to help with your problem we need clear understanding how to reproduce your problem.  
From what you have provided as minimum valuable example and expectation that `ros2 topic info /map --verbose` shall indicate that subscription created from python code shall be with QoS `durability: transient_local` - doesn't relates to the `ros2 bag play` command.

It would be better to ask such questions on https://answers.ros.org/questions/ or in issue trackers for relevant packages `rclpy` and `rclcpp`.

- I am closing this issue as irrelevant to the rosbag2. Feel free to open a new issue with more details if it directly relates to the rosbag2.

closed 2022-12-07 by MichaelOrlov

laurafbec, 2022-12-07:
@MichaelOrlov I don't understand the criteria you have followed to close the issue I posted. I've provided you with a minimal node to record /map topic data with rosbag2_py. I don't have problems with it. The problems appear when using the recorded data with ros2 bag play. QoS profile does not match because it's not properly settled in the function 
` topic_info_map = rosbag2_py._storage.TopicMetadata(name='map', 
                                                          type='nav_msgs/msg/OccupancyGrid', 
                                                          serialization_format='cdr', 
                                                          offered_qos_profiles=' history: keep_all\n  depth: 5\n  reliability: reliable\n  durability: transient_local\n  deadline:\n    sec: 0\n    nsec: 0\n  lifespan:\n    sec: 0\n    nsec: 0\n  liveliness: system_default\n  liveliness_lease_duration:\n    sec: 0\n    nsec: 0\n  avoid_ros_namespace_conventions: false')
        `

no one can tell me how should I establish the QoS profile in the TopicMetada function?

MichaelOrlov, 2022-12-07:
As for me it look like you have a problem from the first hand that your subscription doesn't apply QoS parameters as you provided to it.

MichaelOrlov, 2022-12-08:
As regards to the initialization of the `TopicMetadata::offered_qos_profile` it would be better to use converter function since values to be expected there are integer numbers rather than enum names.
As an example see relevant code for conversion in rqt_bag
https://github.com/ros-visualization/rqt_bag/blob/c1ced6a108a4281e001cccc7feeeac27de68bae8/rqt_bag/src/rqt_bag/qos.py#L58-L73