ros2/rosbag2 #1188: rosbag2_storage_mcap message definition cache cannot find `.msg` definitions outside `msg` dir

james-rms, 2022-11-29:
## Description
Copying over the description from https://github.com/ros-tooling/rosbag2_storage_mcap/pull/77 : 

> We were trying to use mcap on a large ROS2 project and ran into an issue where `.msg` files were not being found in the `install/<package_name>/share/<package_name>/msg/*.msg` path. We got errors like below and the produced mcap files were empty.

```
[node-2] [WARN] [1668788082.046164202] [rosbag2_storage_mcap]: no .msg definition for AAA/BBB.msg, falling back to IDL
```

> On closer look we saw that our package is structured to create `.msg` files in paths like `install/<package_name>/share/<package_name>/<sub_directory_name>/*.msg`. This is purely for organizational purposes and ROS2 doesn't have issue in finding these. So this PR modifies the MCAP plugin code to look for `.msg` files in any sub-directory within `install/<package_name>/share/<package_name>/`. It resolves our problem

### Expected Behavior
The MCAP storage plugin finds MSG definitions in any directory that the rest of ROS2 build infrastructure deems acceptable.

### Actual Behavior
the MCAP storage plugin fails to find MSG definitions outside of an `msg` package subdirectory.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. Logs print error '...' **

## System (please complete the following information)
 - OS: [e.g. Ubuntu Bionic]
 - ROS 2 Distro: [e.g. Dashing]
 - Version: [release, branch, or commit hash]

## Additional context
** Add any other context about the problem here **

jhurliman, 2022-11-30:
This should fix the issue, just needs to be ported to this repo: https://github.com/foxglove/ros-foxglove-bridge/pull/95

aditya2592, 2022-12-19:
@jhurliman Thanks for making the changes in foxglove-bridge, I copied over and tested the changes to this repo [here](https://github.com/ros2/rosbag2/pull/1218). Would be great if you could have a look whenever possible!

MichaelOrlov, 2025-07-02:
- Closing as fixed in the https://github.com/ros2/rosbag2/pull/2055