ros2/rosbag2 #1191: Unable to record in MCAP since last Humble sync

doisyg, 2022-11-30:
## Description
Unable to record in MCAP since last Humble sync

> 
> ros2 bag record --storage mcap /cmd_vel
> [INFO] [1669834334.369219902] [rosbag2_recorder]: Press SPACE for pausing/resuming
> [ERROR] [1669834334.370525009] [rosbag2_storage]: Could not open 'rosbag2_2022_11_30-18_52_14/rosbag2_2022_11_30-18_52_14_0' with 'mcap'. Error: unknown MCAP storage preset profile (valid options are 'fastwrite', 'zstd_fast', 'zstd_small'): none
> [ERROR] [1669834334.370542104] [rosbag2_storage]: Could not load/open plugin with storage id 'mcap'
> No storage could be initialized. Abort

james-rms, 2022-11-30:
Hi @doisyg , thanks for the report. This was previously reported in another repo: https://github.com/ros-tooling/rosbag2_storage_mcap/issues/82 - we're waiting on the next humble sync to ship the fix. I can leave this issue open for visibility until that sync happens.

doisyg, 2022-11-30:
Thanks for the reply @james-rms and the already fixed issue.
Maybe these basic features could be protected by some CI tests ?

james-rms, 2022-11-30:
Yes, that's captured in this issue: https://github.com/ros2/rosbag2/issues/1185

doisyg, 2022-12-22:
Closing as solved with the last sync

closed 2022-12-22 by doisyg