ros2/rosbag2 #1195: Warning when using rosbag2 mcap on humble: No storage plugin found with id 'mcap'

ajayvohra2005, 2022-12-03:
## Description
When I open a rosbag2 recorded with `mcap` in a Python client, I get the warning below:
`[rosbag2_storage]: No storage plugin found with id 'mcap'`

Other than the warning, it appears to work fine and I see no other errors.

### Expected Behavior
Do not expect to see no `[rosbag2_storage]: No storage plugin found with id 'mcap'` warning.

### Actual Behavior
[WARN] [1670089642.967771980] [rosbag2_storage]: No storage plugin found with id 'mcap'

## To Reproduce
** Steps to reproduce the behavior, e.g.
```
>>> import rosbag2_py
>>> bag_path= # set to your rosbag2 directory
>>> reader = rosbag2_py.SequentialReader()
>>> storage_options=rosbag2_py.StorageOptions(uri=bag_path, storage_id='mcap', storage_preset_profile="zstd_fast")
>>> converter_options = rosbag2_py.ConverterOptions(input_serialization_format='cdr',output_serialization_format='cdr')
>>> reader.open(storage_options, converter_options)
[WARN] [1670089642.967771980] [rosbag2_storage]: No storage plugin found with id 'mcap'.
>>>
```

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Humble
 - Version: [release, branch, or commit hash]

## Additional context
** Add any other context about the problem here **

emersonknapp, 2022-12-04:
Have you installed the package `ros-humble-rosbag2-storage-mcap`? This does not ship by default in Humble with the `ros-humble-rosbag2` metapackage.

MichaelOrlov, 2022-12-04:
- It looks like duplicate of the https://github.com/ros2/rosbag2/issues/1191

wep21, 2022-12-07:
https://github.com/ros2/rosbag2/blob/05d2e43b57445d5e2e2db7bd70399c29e335bd55/rosbag2_storage/src/rosbag2_storage/impl/storage_factory_impl.hpp#L183-L201
I guess when opening a read only interface, read-only plugins are explored first, so no storage plugin warnings are always coming up because sqlite3 or mcap is a read write plugin.

MichaelOrlov, 2022-12-07:
Ahh, - if you are just worry about those warning and everything else works as expected. Then yes we have fixed those warning on rolling in the scope of the 
- #1117
- I have created #1201 - Although not sure if we will be able to merge it to Humble since it's adding new virtual method `virtual void load_prev_file();` and it could break ABI compatibility. 
- Anyway as alternative we can just remove those logging warning in https://github.com/ros2/rosbag2/blob/e9c10157095bcbf54186f435e1021ede3d473ec9/rosbag2_storage/src/rosbag2_storage/impl/storage_factory_impl.hpp#L119-L120

emersonknapp, 2022-12-12:
> Anyway as alternative we can just remove those logging warning in

I thing we should just do this - the messaging was always misleading.

emersonknapp, 2023-02-01:
Fixed in #1219

PR #1219 (2022-12-21): Disable false message about plugin not found on bag opening
- Closes #1195 
The error about not found plugin was incorrectly reporting on bag open. Error happened because we are trying to open storage in read only mode first, although SQLite3 an MCAP plugins supports only read-write mode.
closed 2023-02-01 by emersonknapp