ros2/rosbag2 #1212: `ros2 bag record` log file is given  base name `python3....`

russkel, 2022-12-14:
## Description
When launching from a python launch file like so:

```
    ros_bag = ExecuteProcess(
        name="ros2bag",
        cmd=[
            "ros2",
            "bag",
            "record",
            # "-s",
            # "mcap",  # this does not seem to work - may need to wait until next humble sync
            "--all",
            # "--storage-preset-profile",
            # "none",  # this is weird
            "-o",
            PathJoinSubstitution([mission_dir, "bag"]),
        ],
        output="screen",  # ros2bag has its own logging
    )
```

It creates a log file of the filename `python3_3026908_1671001643198.log`.

### Expected Behavior
A more appropriate filename prefixed with `rosbag2_recorder` as per the node name.

## To Reproduce
** Steps to reproduce the behavior, e.g.
Run above in a launch file.
`ls` the log directory.

## System (please complete the following information)
 - OS: jammy
 - ROS 2 Distro: humble
 - Version: 
 ```
ros-humble-rosbag2-compression-zstd/jammy,now 0.15.3-1jammy.20221109.184515 amd64 [installed,automatic]
ros-humble-rosbag2-compression/jammy,now 0.15.3-1jammy.20221109.184119 amd64 [installed,automatic]
ros-humble-rosbag2-cpp/jammy,now 0.15.3-1jammy.20221109.183703 amd64 [installed,automatic]
ros-humble-rosbag2-interfaces/jammy,now 0.15.3-1jammy.20221109.182404 amd64 [installed,automatic]
ros-humble-rosbag2-py/jammy,now 0.15.3-1jammy.20221109.191119 amd64 [installed,automatic]
ros-humble-rosbag2-storage-default-plugins/jammy,now 0.15.3-1jammy.20221109.183024 amd64 [installed,automatic]
ros-humble-rosbag2-storage-mcap/jammy,now 0.5.0-1jammy.20221109.184113 amd64 [installed]
ros-humble-rosbag2-storage/jammy,now 0.15.3-1jammy.20221109.182818 amd64 [installed,automatic]
ros-humble-rosbag2-transport/jammy,now 0.15.3-1jammy.20221109.184745 amd64 [installed,automatic]
ros-humble-rosbag2/jammy,now 0.15.3-1jammy.20221109.192253 amd64 [installed,automatic]
```

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2023-02-01:
@russkel It seems this issue unrelated to the rosbag2. It more relates to the Python launch.  
Actually I would expect to see two log files. One for python launch and one for rosbag2 recorder.  
Are you sure that log for rosbag2 not present or that logs messages related to the rosbag2 recorded in the `python3_3026908_1671001643198.log`  ?
Can you attach log file as example?

russkel, 2023-02-01:
> @russkel It seems this issue unrelated to the rosbag2. It's more relates to the Python launch.

I don't think it is. I originally thought this was the case and I debugged/stepped right through the launch code during launching and it was not creating any log files (because it is set to `screen`).

If I am not mistaken, `name="ros2bag"` would be enough for launch to correctly set the filename. 

```
$ russ @ kalesmoothie in ~ [15:25:39]
↳ rm -rf ~/.ros/log/*
zsh: sure you want to delete more than 100 files in /home/russ/.ros/log [yn]? y

$ russ @ kalesmoothie in ~ [15:26:20]
↳ source /opt/ros/humble/setup.zsh 

$ russ @ kalesmoothie in ~ [15:26:50]
↳ ros2 bag record --all
[INFO] [1675225614.830445834] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1675225614.833532621] [rosbag2_storage]: Opened database 'rosbag2_2023_02_01-15_26_54/rosbag2_2023_02_01-15_26_54_0.db3' for READ_WRITE.
[INFO] [1675225614.833842246] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1675225614.835087643] [rosbag2_recorder]: Listening for topics...
[INFO] [1675225614.836427656] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1675225614.836970741] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1675225619.656144904] [rclcpp]: signal_handler(signum=2)
[INFO] [1675225619.749181886] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1675225619.851948745] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while

$ russ @ kalesmoothie in ~ [15:26:59]
↳ ls ~/.ros/log/                  
 python3_2730488_1675225614806.log

$ russ @ kalesmoothie in ~ [15:27:54]
↳ cat ~/.ros/log/python3_2730488_1675225614806.log 
[INFO] [1675225614.830445834] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1675225614.833532621] [rosbag2_storage]: Opened database 'rosbag2_2023_02_01-15_26_54/rosbag2_2023_02_01-15_26_54_0.db3' for READ_WRITE.
[INFO] [1675225614.833842246] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1675225614.835087643] [rosbag2_recorder]: Listening for topics...
[INFO] [1675225614.836427656] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1675225614.836970741] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1675225619.656144904] [rclcpp]: signal_handler(signum=2)
```

MichaelOrlov, 2023-02-01:
@russkel Thanks for providing details. Now I see.  
I've tried to trace where we first time calling logger and trace down how it is created and haven't found any connection to the python name.  
We first time calling logging in recorder from `Recorder` constructor  https://github.com/ros2/rosbag2/blob/48b28ae087fa21fa2d1c5e4fc6deb9b2ab9f8f04/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L99-L101
and `get_logger()`  corresponds to the newly created node in a few lines upper https://github.com/ros2/rosbag2/blob/48b28ae087fa21fa2d1c5e4fc6deb9b2ab9f8f04/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L81

In theory log name should correspond to the `node_name` variable which is defined as `rosbag2_recorder` by default in header file.

@clalancette Do you have any ideas why log name corresponds to the weird `python3_2730488_1675225614806.log`  rather than expected `rosbag2_recorder_2730488_1675225614806.log` name?

russkel, 2023-02-01:
> In theory log name should correspond to the `node_name` variable which is defined as `rosbag2_recorder` by default in header file.

I got this far and saw it would involve digging through rclcpp and I got scared off. My guess is because of the launching from ros2cli or the use of the pybind stuff causing some problems.

weeshal, 2023-10-18:
This seems to be the case with all nodes launched in python, atleast the ones launched in windows in my use-case (haven't checked in linux). Any pointers or progress on this?

russkel, 2023-10-18:
I think we need to escalate this to `rclpy` or the like. Maybe I will do so.

russkel, 2023-10-22:
This issue has already been reported here: https://github.com/ros2/ros2cli/issues/856

rcutils by default uses process name.

clalancette, 2023-10-22:
Yes, given that we have an issue in the other repository, I'm going to close this one out.

closed 2023-10-22 by clalancette