ros2/rosbag2 #1229: Bad dependencies in rosbag2 ros:rolling apt packages?

hodnajit, 2023-01-09:
## Description
The newest ros-rolling-rosbag2-storage-default-plugins is not installed via apt update && apt upgrade, and it is kept back. With that, ros2 bag play gives me segmentation fault in ros:rolling docker image.

### Expected Behavior
All newest apt packages for rosbag2 are installed.

### Actual Behavior
ros-rolling-rosbag2-storage-default-plugins is kept back and it is not installed.

## To Reproduce
1. run docker image ros:rolling
`docker run -it ros:rolling`
3. call update upgrade and see, that this package is kept back
`apt-get update && apt-get upgrade -y`
5. this package might be installed manually with
`apt install ros-rolling-rosbag2-storage-default-plugins`
and everything works fine, but I think, it should be installed with update && upgrade.

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: ROS:ROLLING
 - Version: release ros-rolling-rosbag2-storage-default-plugins amd64 0.18.0-3jammy.20221229.23380

emersonknapp, 2023-01-11:
> call update upgrade and see, that this package is kept back

Do you have the exact output for this?

I just tried the steps you provided locally (thanks for using docker!) but everything seems to have worked properly `rosbag2-storage-default-plugins` installs 0.18.0 on the upgrade step without manual intervention. Do you get the same result if you try again today?

hodnajit, 2023-01-11:
Thanks for replying. Today these steps work:
1) run docker image
`docker run -it ros:rolling`
2) call apt update && apt upgrade
`apt update && apt upgrade -y`

But, something strange is happening if I put these lines into dockerfile directly.
Dockerfile:
```
FROM ros:rolling

RUN apt-get update && apt-get upgrade -y
```
And with these, the 
```
The following packages have been kept back:
  ros-rolling-rosbag2-storage-default-plugins
```

emersonknapp, 2023-01-12:
I now understand why this is happening.

* `rosbag2_storage_default_plugins` has a new dependency in between the docker base image and the latest Rolling (`rosbag2_storage_sqlite3`). 
* `apt-get upgrade`  will _not_ install new packages by default. See the option `--with-new-pkgs` on https://manpages.debian.org/stretch/apt/apt-get.8.en.html.
* `apt upgrade` _will_ install new packages by default

We were perhaps careless in using `apt` and `apt-get` interchangeably. One is not actually shorthand for another, they are different tools with subtly different semantics.

So, if you instead do 

```
FROM ros:rolling
RUN apt-get update && apt-get upgrade -y --with-new-pkgs
```

OR

```
RUN apt update && apt upgrade -y
```

You will now see instead the output

```
The following NEW packages will be installed:
  ros-rolling-rosbag2-storage-sqlite3
```

However, `apt` is intended to be a user interface, whereas `apt-get` is recommended for automation scripts, so I'll lean towards recommending the former option `--with-new-pkgs` as a general rule for these Dockerfiles.

Please feel free to close the issue if this solves the problem for you!

closed 2023-01-12 by hodnajit