ros2/rosbag2 #1234: Transient Local topics should optionally appear in all bag splits.

vinnnyr, 2023-01-21:
## Description

When creating bag splits, the transient local topics (such as `/tf_static`) that we expect on the system are only written to the first bag of the split series, rather than all of the splits.

While that makes sense, I would like the option to write all transient local topics in all the bags, since I ideally want to be able to use the split bag as a self contained data log, rather than depending on the first bag for information such as `/tf_static`. 

## Related Issues

I am not sure.

## Completion Criteria

- Ros2 bag cli option to optionalally "add transient local data to all bags". 

## Implementation Notes / Suggestions

** If you have ideas about how this feature might be accomplished, put them here. Note that this is just a suggestion to the implementer, so feel free to speculate. **

I am not quite sure yet.

## Testing Notes / Suggestions

** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

Integration Test:
1. publish a transient local topic (idea: tf_static)
2. record a bag with splits enabled, as well as this new feature
3. delete the first bag
4. make sure tf_static is contained in subsequent bags.

emersonknapp, 2023-01-21:
This is a duplicate of https://github.com/ros2/rosbag2/issues/1159

closed 2023-01-21 by emersonknapp

vinnnyr, 2023-01-21:
sorry about that! I didn't recognize it initially.