ros2/rosbag2 #1237: Humble ros2 bag play qos_override doesn't change topic durability

droter, 2023-01-31:
## Description
Original topic is being published as durability: transient_local.  The bag file plays back durability: volatile.  QoS override on playback doesn't change durability. 

Bag Playback with QoS override $ ros2 bag play --qos-profile-overrides-path qos_override.yaml recorded_bag_file

(qos_override.yaml)
/tf_static:
  history: keep_last
  depth: 10
  reliability: reliable
  durability: **transient_local**
 

### Expected Behavior
$ ros2 topic info /tf_static --verbose

Node name: rosbag2_player
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Endpoint type: PUBLISHER
GID: 01.0f.e4.30.37.36.9b.53.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: **TRANSIENT_LOCAL**
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite  

### Actual Behavior
Durability is unchanged.

$ ros2 topic info /tf_static --verbose

Node name: rosbag2_player
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Endpoint type: PUBLISHER
GID: 01.0f.e4.30.37.36.9b.53.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: **VOLATILE**
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite  

## To Reproduce
** Steps to reproduce the behavior, e.g.
Pick topic that is being published as durability: transient_local and record it.
Play the bag back and check the durability.
Play the bag back using qos_override and check the durability.

## System (please complete the following information)
 - OS: Ubuntu on Mac
 - ROS 2 Distro: Humble
 - Version: latest

## Additional context
** Add any other context about the problem here **

emersonknapp, 2023-02-01:
I assume the extra "**" are your way of adding emphasis, and not your literal input?

droter, 2023-02-01:
Yes, just to make it easier to see the parameter that I am trying to change.

emersonknapp, 2023-02-01:
Does `metadata.yaml` list the topic's metadata as `transient_local`? I wonder if it's _also_ being recorded incorrectly, or if the issue is just in publisher creation for playback.

droter, 2023-02-01:
It doesn't show the durability setting.  I would like to record it at the publisher QoS and play it back at the same.

droter, 2023-02-01:
This is the metadata:
    - topic_metadata:
        name: tf_static
        type: tf2_msgs/msg/TFMessage
        serialization_format: cdr
        offered_qos_profiles: ""
      message_count: 1

emersonknapp, 2023-02-01:
Friendly hint: you can use triple-backticks "```" to format code/output blocks - github has full markdown rendering support.

I'm unable to reproduce that bag - how exactly did you record the data in the first place? The empty `offered_qos_profiles`  is suspicious, that's not a valid value given that a message was received. Have you edited the metadata by hand, or is that what came out of your recording process?

Here's my results on a clean latest Humble install:
```
# One shell
$ ros2 run tf2_ros static_transform_publisher --x 1 --y 2 --z 3 --roll 90 --pitch 180 --yaw 270 --frame-id world --child-frame-id base

# Another shell
$ ros2 topic pub -r 3 /chatter std_msgs/String "data: hello"

# Third shell
$ ros2 bag record /tf_static /chatter -o testbag

# ctrl-c after a few seconds
$ cat testbag/metadata.yaml 
rosbag2_bagfile_information:
  version: 5
  storage_identifier: sqlite3
  duration:
    nanoseconds: 6666808227
  starting_time:
    nanoseconds_since_epoch: 1675282923350705648
  message_count: 22
  topics_with_message_count:
    - topic_metadata:
        name: /chatter
        type: std_msgs/msg/String
        serialization_format: cdr
        offered_qos_profiles: "- history: 3\n  depth: 0\n  reliability: 1\n  durability: 1\n  deadline:\n    sec: 9223372036\n    nsec: 854775807\n  lifespan:\n    sec: 9223372036\n    nsec: 854775807\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 9223372036\n    nsec: 854775807\n  avoid_ros_namespace_conventions: false"
      message_count: 21
    - topic_metadata:
        name: /tf_static
        type: tf2_msgs/msg/TFMessage
        serialization_format: cdr
        offered_qos_profiles: "- history: 3\n  depth: 0\n  reliability: 1\n  durability: 1\n  deadline:\n    sec: 9223372036\n    nsec: 854775807\n  lifespan:\n    sec: 9223372036\n    nsec: 854775807\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 9223372036\n    nsec: 854775807\n  avoid_ros_namespace_conventions: false"
      message_count: 1
  compression_format: ""
  compression_mode: ""
  relative_file_paths:
    - testbag_0.db3
  files:
    - path: testbag_0.db3
      starting_time:
        nanoseconds_since_epoch: 1675282923350705648
      duration:
        nanoseconds: 6666808227
```

The metadata is showing me the correct information (TRANSIENT_LOCAL's enum value is 1), and when I play back:

```
# one shell
$ ros2 bag play testbag

# another shell while it's still playing
$ ros2 topic info -v /tf_static
Type: tf2_msgs/msg/TFMessage

Publisher count: 1

Node name: rosbag2_player
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Endpoint type: PUBLISHER
GID: 01.0f.eb.7d.22.01.1e.3c.01.00.00.00.00.00.13.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 0
```

However, if I go edit the `metadata.yaml` to get `offered_qos_profiles: ""` and play back, then indeed it does default to `VOLATILE`.

So I create an overrides file:

```
# qos_override.yaml
/tf_static:
  history: keep_last
  depth: 10
  durability: transient_local
```

And play it back

```
$ ros2 bag play --qos-profile-overrides-path qos_override.yaml testbag
[INFO] [1675283765.066072454] [rosbag2_player]: Overriding QoS profile for topic /tf_static

# other shell
$ ros2 topic info -v /tf_static
Type: tf2_msgs/msg/TFMessage

Publisher count: 1

Node name: rosbag2_player
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Endpoint type: PUBLISHER
GID: 01.0f.eb.7d.19.03.9e.25.01.00.00.00.00.00.13.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 0
```

I see the override applied successfully.

Are you able to reproduce your issue by following steps like the above? If not, are you able to determine what the difference is between what I've presented and the steps that you are taking?

droter, 2023-02-01:
Thanks for the testing ideas.  I ran through and was able to reproduce the results you have. 
[recordbag.zip](https://github.com/ros2/rosbag2/files/105618
[record.py.txt](https://github.com/ros2/rosbag2/files/10561850/record.py.txt)
37/recordbag.zip)

Attached is the file (record.py) using ros2bag_py.  It doesn't output the offered_qos_profiles in the metadata.  Also, using ros2 bag play override doesn't change the qos when replaying the file. 

If I copy the offered_qos_profiles from your test and paste it into the recordbag metadata.yaml file then it will replay the bagfile with the correct settings.
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Maybe the fix is to get the right data into the metadata.yaml file.

emersonknapp, 2023-02-01:
I do have a question - why are you creating a manual writer instead of using the Recorder that it provides? Not that you _can't_ do this, it's a fully supported workflow, but I see no reason to reinvent the wheels that are already present. You can `rosbag2_py.Recorder` at your disposal. Check out `ros2bag/ros2bag/verb/record.py` as a good example of how it can be used.

Regardless, like you alluded to, your first issue is that since you are manually creating `TopicMetadata`, and omitting `offered_qos_profiles`, that is why the created metadata has no info. This field is filled by the `Recorder` normally via info from discovered publishers. So that explains the recording issue, the API is working as designed.

> Also, using ros2 bag play override doesn't change the qos when replaying the file.

Can you be more specific about this? Did you run through my exact steps above and experience a different result? Can you copy-paste, verbatim, every command that you ran?

droter, 2023-02-01:
I will checkout the rosba2_py.Recorder

In the test you provided the override works.  

When I try the override with my bag recorded with the manual writer it doesn't work.
$ ros2 bag play -l --qos-profile-overrides-path qos_override.yaml recordbag

emersonknapp, 2023-02-01:
So here's the issue: in your topic metadata you've called the topic `"tf_static"` (no namespace), but in your `qos_override.yaml` you're overriding the topic `/tf_static` (with a slash, global namespace). The strings don't match exactly, so  the topic's qos profile is not being overridden.

```
# qos_override.yaml
tf_static:
  history: keep_last
  depth: 10
  durability: transient_local
```

You can solve your issue by minding the exact string used to specify the topic name.

I'm trying to decide if this should be considered a bug, if a topic with or without a leading slash should be considered the same for this purpose

droter, 2023-02-01:
wow  Thanks for looking into this.  I didn't know what was causing the issue.

Yes, the override works with the correct yaml file.

emersonknapp, 2023-02-01:
I am leaning toward considering this a subtle bug in that QoS overrides currently required an exactly-matching topic name. A fully-qualified topic name always starts with a slash, and  `ros2 bag record` only creates fully-qualified topic names in bags, and current testing only used fully-qualified topic names for overrides. But, if a bag contains `my_topic` OR `/my_topic`, a QoS override specifying either `my_topic` OR `/my_topic` should also work, for any of the combinations. 

Note https://design.ros2.org/articles/topic_and_service_names.html#ros-2-topic-and-service-name-constraints for design on ROS 2 names

emersonknapp, 2023-02-01:
Closing this as a completed discussion, to be followed up in #1240

closed 2023-02-01 by emersonknapp