ros2/rosbag2 #1240: Allow QoS overrides for underqualified topic names

emersonknapp, 2023-02-01:
## Description

Followup from #1237 

QoS overrides for bag playback require the override topic name to exactly match the string topic name in the bag. However, ROS 2 names allow for different levels of qualifications, e.g. `my_topic`, and `/my_topic` are different exact strings, but expand to the same fully-qualified topic name `/my_topic`.

rosbag2 should compare only expanded topic names so that semantically-meaningless distinctions in user input are ignored.

https://design.ros2.org/articles/topic_and_service_names.html#ros-2-topic-and-service-name-constraints

## Completion Criteria

If a bag contains `/my_topic` and the QoS override specifies `my_topic`, the override should work. Likewise if the bag contains `my_topic`, then QoS override `/my_topic` should also work.

## Implementation Notes / Suggestions

Fully expand topic names before doing comparisons.

Maybe rosbag2 should fully expand all topic names when creating topics in storage?

## Testing Notes / Suggestions

Add a test for the new cases.

lakhmanisahil, 2026-01-29:
> [@lakhmanisahil](https://github.com/lakhmanisahil) by all means give it a shot!

Thank You @kscottz, for assigning this to me.
I am looking into it, and will aim to shoot a PR!.

lakhmanisahil, 2026-04-07:
Hello @kscottz ,

Apologies for the delay — I was assigned in January but missed following up earlier.

I was able to reproduce the issue:

When a bag stores /talker but the QoS override YAML uses talker: (without a leading slash), the override is silently ignored — no "Overriding QoS profile" log appears, and the player falls back to the recorded QoS.
Changing the YAML key to /talker: makes the override work immediately.

This confirms the issue is due to a direct string comparison — semantically identical topic names with different qualification levels do not match.

I implemented a fix on Humble (0.15.16, where source and apt are aligned). The fix normalizes topic names to fully qualified form (prepending / if absent) on both sides before performing the QoS override map lookup (in both player and recorder).

If this approach looks reasonable, I can open a draft PR for Humble.