ros2/rosbag2 #1252: `rosbag2_storage::SerializedBagMessage` does not contain any publisher information

jhurliman, 2023-02-27:
## Description

In ROS 1 bag files (rosbag1 v2), there is a `callerid` string field for each channel that provides the name of the ROS node that is publishing the channel. In rosbag2, the `SerializedBagMessage` struct only contains a `topic` string without identifying which channel or publisher a message came from. As a result, the MCAP storage plugin creates a single channel per topic instead of a separate channel per publisher to a given topic.

### Expected Behavior

`SerializedBagMessage` contains a `callerid` string or some unique identifier for a publishing channel or a publishing node. The MCAP storage plugin uses this to create separate channels per publisher instead of one channel per topic.

### Actual Behavior

The MCAP storage plugin creates a single channel per topic since `topic` is the only source identifier in `SerializedBagMessage`.

MichaelOrlov, 2023-02-28:
@jhurliman The question is what would be the `publisher identifier` ?
@emersonknapp Any thoughts or suggestions about it?

emersonknapp, 2023-02-28:
In ROS 2, publishers do have an assigned GID. See https://github.com/ros2/rosbag2/compare/rolling...emersonknapp/publisher_gid for where we can get it.

This is very much related to https://github.com/ros2/rosbag2/issues/629 as well. I'm going to reword this issue a little bit. It's a feature request, not a bug, but it describes the foundation for solving #629 as well as the conversation we had on https://github.com/ros2/rosbag2/issues/1169