ros2/rosbag2 #1254: Is there any way to have a higher rate and a deterministic playback with a rosbag file?

KuenstlicheIntelligenz, 2023-03-02:
## Description
Hello.

I try to parse a rosbag with image data taken from a robot. I wrote a node that listens to the published topics and processes the image data.
I play back the rosbag and set the rate to a higher level. 

I tried to set the size of any queue that is involved to a way higher value than the published topics.
(Subscriber queues and the read ahead option of the ros2 bag play)

I set the qos of any related topic to:

```
  history: keep_all
  reliability: reliable
```

### Expected Behavior
Deterministic playback and playback of every single event regardless of the rate.

### Actual Behavior
The rate has an impact of how many messages get processed.

###
I guess there must be a reliable way to playback the whole bag regardless of the rate.
Which parameters could I have forgotten?

MichaelOrlov, 2023-03-02:
@KuenstlicheIntelligenz Deterministic replay is a whole different topic which rosbag2 doesn't support our of the box.
However changing QoS settings 
```
history: keep_last
qos_depth: 15
```
might help in your case.
Also using shared memory transport.

KuenstlicheIntelligenz, 2023-03-08:
Thank you. I've switched to reading the bag in code.

closed 2023-03-08 by KuenstlicheIntelligenz