ros2/rosbag2 #1256: Consistent recording of /clock from replaying a bag

Grisly00, 2023-03-06:
## Description
I try to play a mcap bag file and simultaneously record a new bag file while converting some messages. (Ouster lidar_packets to Pointcloud2). Everything is started from a launch file. I play the bag file with the following command:
```bash
ros2 bag play input_bagfile --rate 0.5 --clock --disable-keyboard-controls --read-ahead-queue-size 10000 --start-paused
```
and record with the following:

```bash
ros2 bag record -s mcap --qos-profile-overrides-path qos_profile_override.yaml --use-sim-time --max-bag-size 10485760000 -a
```

Sometimes publishing /clock from one bag file to the other works fine, but other times the starting time of the newly recorded bag file is set to `Jan  1 1970 01:00:00.0 (0.0)`. I guess this happens when the recording is started but /clock is not (yet) published.
I tried to start the recording with a delay to make sure that the bag is playing, but this didn't help (or at least didn't completely avoid the problem). I guess it comes down to qos settings for /clock, but as far as I can tell they cant be changed.

Does anyone have a suggestion what I can do to get a reliable outcome when converting bag files?

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Humble

Grisly00, 2023-03-10:
@MichaelOrlov Unfortunately not it seems. I need to translate one topic into another one while converting. 
In the input bag I have raw data from a Ouster lidar which are called lidar_packets and I need to transform them into Pointcloud2 messages. Therefore I have the Ouster ROS2 driver running while I a play and record the bag.
This means I start the Ouster driver, then I play the bag file with the raw data and then I start the recording of the translated package.
As far as I understand, this is not possible with `ros2 bag convert`.

adamkrawczyk, 2023-04-17:
@Grisly00 Unfortunately I have no idea how to solve this issue, check the simulator to see if the /clock is published correctly and you can remove "malformed" messages with the database browser (ex. DB browser)

emersonknapp, 2023-05-26:
For this case, I might suggest just writing a script/program to do the conversion you want, instead of depending on transport. Read the bag one message at a time, call the conversion function, and write it out to a new bag. That's going to be a lot more reliable and performant than depending on publishing messages from one bag to another with the uncertainty that is inherent in discovery and network transport.

MichaelOrlov, 2023-05-29:
@Grisly00 In our case I would recommend to use `--clock-topics-all` parameter for player instead of  simple `--clock`
```
'--clock-topics-all', default=False, action='store_true',
            help='Publishes an update on /clock immediately before each replayed message'
```
At least in this case, you will be sure that "clock" messages will be always published before messages from the topic.

MichaelOrlov, 2023-06-01:
@Grisly00 It is unlikely will be possible o make a backport of this feature to the Humble due to the requirement to keep API/ABI stability for the ROS2 released distros.

As regards the :
>is the rolling branch compatible with a ros2 humble installation?

If consider just rosbag2 repo - there is no guarantee that rosbag2 from Rolling will be fully functional with the rest Humble release. However, we keep backward compatibility in the rosbag2 player. i.e. bags recorded in previous distro e.g. Humble or Foxy will be playable in rosbag2 player from Rolling.

You can give it a try or backport #1036, #1050 to Humble and compile it from sources.

MichaelOrlov, 2023-06-01:
- Closing for now as a workaround with existing functionality `--clock-topics-all` parameter has been found.
Feel free to reopen if decided that we can do anything else in addition to already provided workaround.

closed 2023-06-01 by MichaelOrlov

Grisly00, 2023-06-07:
Thank you for the infos. I tried to test the `--clock-topics-all` flag in a rolling installation, but had issues with the Ouster lidar driver so I can not confirm if it solves the issue yet. 

> @Grisly00 It is unlikely will be possible o make a backport of this feature to the Humble due to the requirement to keep API/ABI stability for the ROS2 released distros.

I understand that it is important to not mess up the API, but considering that this is potentially quite braking I would really like to see these things in Humble, after all it is the newest LTS distro. (On a related note this issue #841 gave me a major headache until I found out whats going on, as it didn't get ported to Humble.)

clalancette, 2023-06-07:
> I understand that it is important to not mess up the API, but considering that this is potentially quite braking I would really like to see these things in Humble, after all it is the newest LTS [...]

There is usually a way to do the backport while maintaining API/ABI, though it involves some creative thinking and someone with the time to invest in it.

MichaelOrlov, 2023-06-07:
@Grisly00 Could you please clarify on 
> On a related note this issue https://github.com/ros2/rosbag2/issues/841 gave me a major headache until I found out whats going on, as it didn't get ported to Humble.

- As far as I understand the #841 was not affecting Humble since the source code is different and a potential fix is already there.
See my comment in https://github.com/ros2/rosbag2/pull/1101#pullrequestreview-1380241650. Do you think or observe that this is not true?

Grisly00, 2023-06-09:
`There is usually a way to do the backport while maintaining API/ABI, though it involves some creative thinking and someone with the time to invest in it.`
I of course understand that things take time, I should have been more clear. It was really meant as a question if new features only get implemented in Rolling or if the general plan is to also have the latest features and fixes in Humble.

 `As far as I understand the Bag split updating starting_time metadata field  #841 was not affecting Humble since the source code is different and a potential fix is already there.
See my comment in https://github.com/ros2/rosbag2/pull/1101#pullrequestreview-1380241650. Do you think or observe that this is not true?`
Sorry for mixing Issues here, but yes I do experience this on Humble as well and did not see your comment.

MichaelOrlov, 2023-06-13:
@Grisly00 As regards
> It was really meant as a question if new features are only implemented in Rolling or if the general plan is to have the latest features and fixes in Humble as well.

Usually, we do the development of the new features on the Rolling and then trying to backport them to the other ROS2 distros which have not yet reached the end of life.  
If backport is trivial - it will be made almost automatically by `mergify` bot. If it has conflicts, other dependencies or API/ABI incompatible changes we will do our best to backport them but there are no guarantees in this case.

If one needs some particular feature to be backported help with backporting will be very appreciated.

Grisly00, 2023-06-14:
@MichaelOrlov Thank you for the clarification. Concerning the issue with the metadata file, I will observe if it continues to occur in Humble, and if so, I will open an issue.