ros2/rosbag2 #1258: Version Documentation?

DLu, 2023-03-08:
## Description

Is there an actual version spec somewhere that I can't find? I'm trying to figure out the redundancy between `relative_file_paths` and `files/path`, and I see different versions [referenced in the code](https://github.com/ros2/rosbag2/blob/fa60ff77b600010023001e373bf95723db71165a/rosbag2_compression/src/rosbag2_compression/sequential_compression_reader.cpp#L63) but my preliminary searches have not uncovered documentation. 

## Completion Criteria

 * A description of what fields go in the metadata in the different versions. 

## Implementation Notes / Suggestions

Markdown in github or docs.ros info.

MichaelOrlov, 2023-03-09:
@DLu Sorry no documentation for that.
Please refer to the YAML parser decode function
https://github.com/ros2/rosbag2/blob/ff2f057625b0ccc4e56175e6a0f61be00c2d8e76/rosbag2_storage/include/rosbag2_storage/yaml.hpp#L244-L271
for up to date information about what fields go in the metadata in the different versions.

MichaelOrlov, 2023-03-17:
- Closing as answer on question was found.

closed 2023-03-17 by MichaelOrlov