ros2/rosbag2 #1262: Memory error when convert bagfile with multiple writers and compression(message mode) is required

zeal-up, 2023-03-15:
## Description
I am implementing a program to extract bag file to multiple compression output。I wrote the code refer to [rosbag2/bag_rewrite.cpp](https://github.com/ros2/rosbag2/blob/humble/rosbag2_transport/src/rosbag2_transport/bag_rewrite.cpp)。
Nearly all situations the rosbag2 api works fine, but when multiple writer is created with compression_mode=message, the program will raise a memory error.

I than reproduce the error with rosbag2 cli tool.

When convert a bagfile with a output_options.yaml as described in README，if multiple output bag file is required and compression mode is zstd with message level compression, rosbag2_converter will raise a coredump.
output_options.yaml:
```yaml
output_bags:
- uri: out_bag01
  all: true
  compression_format: zstd
  compression_mode: message
- uri: out_bag02
  all: true
  compression_format: zstd
  compression_mode: message
```

### Expected Behavior
Convert bagfile to multiple bagfile with compression.

### Actual Behavior
Different core information will raise in different run, looks like the follows
```
malloc(): unsorted double linked list corrupted
free(): corrupted unsorted chunks
corrupted double-linked list
```

## To Reproduce
** Steps to reproduce the behavior
1. download or generate ros2 bagfile, 
2. touch a file name ouput_options.yaml with follow contens
```
output_bags:
- uri: out_bag01
  all: true
  compression_format: zstd
  compression_mode: message
- uri: out_bag02
  all: true
  compression_format: zstd
  compression_mode: message
```
3. excute
```
ros2 bag convert -i ./path/to/bag/file -o output_options.yaml
```

## System (please complete the following information)
 - OS: Ubuntu 22.04
 - ROS 2 Distro: humble
 - Version: release

## Additional context
I actually found part of the problem.
1. In gdb

[image]

The gdb shows that the error point is when the ZstdCompressor call `rcutils_uint8_array_resize` which will operate the uint8 arrary own by the [SerializedMessage](https://github.com/ros2/rosbag2/blob/14bf3dcc2170ebe5bd73ed0cf83bb4077b3e36b4/rosbag2_transport/src/rosbag2_transport/bag_rewrite.cpp#L151)
2. If no compression is required, everything is fine even multiple writers are writing together
3. I check the different between normal [sequential_writer](https://github.com/ros2/rosbag2/blob/humble/rosbag2_cpp/src/rosbag2_cpp/writers/sequential_writer.cpp) and [sequential_compression_writer](https://github.com/ros2/rosbag2/blob/humble/rosbag2_compression/src/rosbag2_compression/sequential_compression_writer.cpp), the different is that the compression_writer will push the message into a [compressor_message_queue_](https://github.com/ros2/rosbag2/blob/14bf3dcc2170ebe5bd73ed0cf83bb4077b3e36b4/rosbag2_compression/src/rosbag2_compression/sequential_compression_writer.cpp#L320)
4. The `compressor_message_queue_` will process in another thread
5. since the SerializedMessage is built as a shared_ptr, if multiple writer with compression is writing a SerializedMessageSharedPtr, multiple thread will operate the save memory as the same time

## Fix
I fix the bug by deepcopy a SerializedMessage when calling writer->write(SerializedMessageSharedPtr), the code looks like follow:
```
for (auto & [errorTimeInterval, writer] : outputWriters)
        {
            /**
             * 执行拷贝
             * 这里需要拷贝的情景是在有多个writer，并且writer执行压缩时会发生bug，主要的原因应该时在zstd的压缩器会保留一个队列进行压缩，
             * 如果用同一块地址的数据给多个writer的时候，不同的writer执行压缩的时候
             * */ 
            std::shared_ptr<rosbag2_storage::SerializedBagMessage> nextMsg_copy = std::make_shared<rosbag2_storage::SerializedBagMessage>();
            nextMsg_copy->serialized_data = std::shared_ptr<rcutils_uint8_array_t>(
            new rcutils_uint8_array_t,
            [](rcutils_uint8_array_t * msg) {
            auto fini_return = rcutils_uint8_array_fini(msg);
            delete msg;
            if (fini_return != RCUTILS_RET_OK) {
                RCLCPP_ERROR_STREAM(
                rclcpp::get_logger("rosbag2_cpp"),
                "Failed to destroy serialized message: " << rcutils_get_error_string().str);
            }
            });
            rcutils_allocator_t allocator = rcutils_get_default_allocator();
            auto ret = rcutils_uint8_array_init(
                nextMsg_copy->serialized_data.get(),
                nextMsg->serialized_data->buffer_capacity,
                &allocator);
            if (ret != RCUTILS_RET_OK)
            {
                RCLCPP_ERROR(get_logger(), "Init arrry failed with ret code = %d", ret);
            }
            // RCLCPP_INFO(get_logger(), "Resized");
            nextMsg_copy->topic_name = nextMsg->topic_name;
            nextMsg_copy->time_stamp = nextMsg->time_stamp;
            nextMsg_copy->serialized_data->buffer_capacity = nextMsg->serialized_data->buffer_capacity;
            nextMsg_copy->serialized_data->buffer_length = nextMsg->serialized_data->buffer_length;
            // RCLCPP_INFO(get_logger(), "Set length");
            memcpy((void*)nextMsg_copy->serialized_data->buffer,
                    (void*)nextMsg->serialized_data->buffer, nextMsg->serialized_data->buffer_length);
            // RCLCPP_INFO(get_logger(), "Copy");
            // -------------------------------------------------------------------------------------------

            writer->write(nextMsg);
        // RCLCPP_INFO_THROTTLE(get_logger(), *get_clock(), 1000, "writing...");
        RCLCPP_INFO(get_logger(), "Writing...");
        }
```

zeal-up, 2023-03-15:
I wonder to know if there is a more elegant solution?
I am happy to pull a request ~

MichaelOrlov, 2023-03-17:
Hi @zeal-up. The issue which you described should have already been fixed on `Rolling` branch as part of the #1010.
Unfortunately we can't backport #1010 to `Humble` since there are breaking API/ABI changes.

The main problem in backporting that we can't change API for 
https://github.com/ros2/rosbag2/blob/76665ad8b977721c059b95d307697b8424fed07d/rosbag2_compression/include/rosbag2_compression/base_compressor_interface.hpp#L65-L67
By adding pointer to the output `compressed_message`.

zeal-up, 2023-03-30:
Thanks for your reply @MichaelOrlov . Yes, the issue I described is fixed in [#1010 ](https://github.com/ros2/rosbag2/pull/1010). But can't we fix the `ros2 bag convert` bug in [rosbag2/bag_rewrite.cpp](https://github.com/ros2/rosbag2/blob/humble/rosbag2_transport/src/rosbag2_transport/bag_rewrite.cpp)? For example, by a little tricky way I show in the issue.

MichaelOrlov, 2023-03-30:
In theory yes we can add extra copy in [bag_rewrite.cpp](https://github.com/ros2/rosbag2/blob/humble/rosbag2_transport/src/rosbag2_transport/bag_rewrite.cpp) as you made in your own converter. But with condition that it will be only on Humble since we can't propgate more proper fix from #1010.

If you will be able to prepare PR targeting Humble branch it will be much appreciated.

zeal-up, 2023-05-14:
@MichaelOrlov Hi, I prepare a [PR](https://github.com/ros2/rosbag2/pull/1331#issue-1708860338) to fix the reported bug. The PR will only fix the bug in Humble.

MichaelOrlov, 2024-05-05:
- Closing as fixed in the #1331

closed 2024-05-05 by MichaelOrlov