ros2/rosbag2 #1263: Question: record rosbag will all topics excluding a certain list of topics

AswinkarthikeyenAK, 2023-03-15:
Hi, 
I am using ROS humble and was wondering which branch or commit should I have to use to record a rosbag file which can record all the topics however exclude a few topics. 

ros2 bag record -s mcap --all --exclude /hi -> this works fine, 
however, if add another topic after /hi, I get an error saying "[ERROR] [ros2bag]: Must specify only one option out of topics, --regex or --all"

I noticed there were few discussion around this and maybe it has already been fixed, however, it is quite hard to track down which branch should be used. 
Can someone point  me to the correct branch or documentation, please?

Thanks

emersonknapp, 2023-03-16:
I see what is happening, it's a little confusing. The `--exclude` option only takes a single argument, not a list. It can take a regular expression. So your command is being parsed as, "all, exclude /hi, record topic /other topic", the extra topic you provided is going into the positional argument which is a list of topics to record. The "all" and "list of topics" options are not compatible to avoid confusion, so that's where the error comes from.

It would probably make sense to allow for several arguments to "exclude" (PRs welcome!) but for now you should be able to achieve the goal with `--exclude "(/foo|/bar) "` using the regex OR operator

AswinkarthikeyenAK, 2023-03-22:
Hi,
Thanks for the response, it worked. 

I have another question. I am trying to throttle multiple messages (able to list and hz the _throttle topics), however, while trying to ros2 bag record some messages, --excluding the original topics so I can save the throttle topics, this does not save the _throttle messages. 

For instance, in three terminals I run, 

- $ ros2 topic pub -r 10 /hi std_msgs/msg/String "{data: 'Hello from terminal'}"
- $ ros2 topic pub -r 10 /welcome std_msgs/msg/String "{data: 'Hello from terminal'}"
- $ ros2 topic pub -r 10 /greetings std_msgs/msg/String "{data: 'Hello from terminal'}"

and the throttle these messages, I notice 

- /greetings
- /greetings_throttle
- /hi
- /parameter_events
- /rosout
- /welcome
- /welcome_throttle

lets say, I would like to save /greetings_throttle,/parameter_events,/rosout,/welcome_throttle, /hi
I used the following, 
$ ros2 bag record -s mcap --all --exclude "(/welcome|/greetings)"

this resulted in the following:

- [INFO] [1679483609.096532192] [rosbag2_recorder]: Press SPACE for pausing/resuming
- [INFO] [1679483609.097132760] [rosbag2_recorder]: Event publisher thread: Starting
- [INFO] [1679483609.097399096] [rosbag2_recorder]: Listening for topics...
- [INFO] [1679483609.098440541] [rosbag2_recorder]: Subscribed to topic '/rosout'
- [INFO] [1679483609.099232882] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
- [INFO] [1679483609.099659888] [rosbag2_recorder]: Subscribed to topic '/hi'
- [INFO] [1679483609.099906871] [rosbag2_recorder]: Subscribed to topic '/events/write_split'

leaving out the /greetings_throttle and /welcome_throttle.

On the other hand, 
$ ros2 bag record -s mcap --all 

recorded all the topics. Please look below. 

- [INFO] [1679483749.937387135] [rosbag2_recorder]: Press SPACE for pausing/resuming
- [INFO] [1679483749.937981694] [rosbag2_recorder]: Event publisher thread: Starting
- [INFO] [1679483749.938239531] [rosbag2_recorder]: Listening for topics...
- [INFO] [1679483749.938859921] [rosbag2_recorder]: Subscribed to topic '/welcome_throttle'
- [INFO] [1679483749.939526887] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
- [INFO] [1679483749.939726297] [rosbag2_recorder]: Subscribed to topic '/hi'
- [INFO] [1679483749.939903405] [rosbag2_recorder]: Subscribed to topic '/greetings_throttle'
- [INFO] [1679483749.940080629] [rosbag2_recorder]: Subscribed to topic '/welcome'
- [INFO] [1679483749.940748939] [rosbag2_recorder]: Subscribed to topic '/rosout'
- [INFO] [1679483749.940950874] [rosbag2_recorder]: Subscribed to topic '/greetings'
- [INFO] [1679483749.941175776] [rosbag2_recorder]: Subscribed to topic '/events/write_split'

Any reason why this is happening and is there a solution/workaround for this issue?

Thanks

banisadr, 2023-03-30:
Hey @AswinkarthikeyenAK I think you're running into a regex issue. Your `(/foo|/bar) pattern is probably partially matching both the `/foo` and the `/foo_throttled`. I think if you specify the string end parameter, you'll avoid the partial match on the `_throttled` messages you're trying to keep: `--exclude "(/foo$|/bar$)"`.

Hope that helps. If that's not quite it, I'd recommend poking around with the regex a bit to make sure you're not matching topics you want to keep.

AswinkarthikeyenAK, 2023-04-04:
Hey @banisadr, I used regex to subscribe to the topics I wanted. I believe the same, the partial match causes the topics to be excluded as well. Thanks for your suggestion.

closed 2023-04-04 by AswinkarthikeyenAK