ros2/rosbag2 #1282: Test failure `rosbag2_cpp/test_time_controller_clock` on nightly_win_rep

Yadunund, 2023-04-12:
CI: https://ci.ros2.org/view/nightly/job/nightly_win_rep/2955/testReport/(root)/projectroot/test_time_controller_clock/

```
C:\ci\ws\src\ros2\rosbag2\rosbag2_cpp\test\rosbag2_cpp\test_time_controller_clock.cpp(168): error: Value of: end - start >= std::chrono::seconds(1)
  Actual: false
Expected: true
C:\ci\ws\src\ros2\rosbag2\rosbag2_cpp\test\rosbag2_cpp\test_time_controller_clock.cpp(169): error: Value of: sleep_result
  Actual: false
Expected: true
[  FAILED  ] TimeControllerClockTest.unpaused_sleep_returns_true (999 ms)
```

MichaelOrlov, 2023-04-13:
This is flaky test - addressing in https://github.com/ros2/rosbag2/pull/1290

emersonknapp, 2023-06-07:
I think I just ran into this again - I'm not sure it's actually fixed.

Windows running just `test_time_controller_clock` with `--retest-until-failed 100` [image]

emersonknapp, 2023-06-07:
https://github.com/ros2/rosbag2/pull/1367 also is investigating this - I am reopening the issue while investigation is ongoing

MichaelOrlov, 2023-06-07:
@emersonknapp It seems it failed with a different error message on your CI job.
Please refer to my post here https://github.com/ros2/rosbag2/pull/1371#issuecomment-1581652658 for details.