ros2/rosbag2 #1286: Recording hidden action topics fails with `-s mcap`, works with default storage plugin

achim-k, 2023-04-12:
## Description
Recording hidden action topic with `-s mcap` fails with `ValueError: Invalid package resource name` error.

Example failure:
```
root@daa5920a7ce8:/# ros2 bag record -s mcap --include-hidden-topics /fibonacci/_action/feedback
[INFO] [1681311592.569532135] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1681311592.570553808] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1681311592.571091987] [rosbag2_recorder]: Listening for topics...
[INFO] [1681311592.571537769] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1681311592.661370658] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.5', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2bag/command/bag.py", line 38, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2bag/verb/record.py", line 245, in main
    recorder.record(storage_options, record_options)
ValueError: Invalid package resource name: action_tutorials_interfaces/action/Fibonacci_FeedbackMessage
```

### Expected Behavior
Recording hidden action topics should not fail, as is the case when the default storage plugin is used.

### Actual Behavior
** clear and concise description of what _actually_ happened. include copied logs whenever possible **

## To Reproduce

Easiest is probably to run the following commands in a docker container (`docker run -it --rm ros:humble-ros-base`):

```bash
sudo apt-get update \
  && sudo apt-get install -y \
    ros-$ROS_DISTRO-action-tutorials-cpp \
    ros-$ROS_DISTRO-rosbag2-storage-mcap

# launch action server
ros2 run action_tutorials_cpp fibonacci_action_server &

# confirm that the action server is running
ros2 action list

# confirm that "recording" works with default rosbag storage plugin:
ros2 bag record --include-hidden-topics /fibonacci/_action/feedback

# confirm that it doesn't work with mcap storage plugin:
ros2 bag record -s mcap --include-hidden-topics /fibonacci/_action/feedback
```

## System (please complete the following information)
 - OS: `Ubuntu 22.04.2 LTS` (latest `ros:humble-ros-base` docker image)
 - ROS 2 Distro: `humble`
 - Version: `release` (Installed: 0.15.4-2jammy.20230302.215223)

MichaelOrlov, 2023-04-13:
Preliminary analysis:
The exception coming from here
https://github.com/ros2/rosbag2/blob/9d64382f833ce211b9841ef8e47047d3d5099d64/rosbag2_storage_mcap/src/message_definition_cache.cpp#L136-L141
it looks like `std::regex_match` fails to match resource name `action_tutorials_interfaces/action/Fibonacci_FeedbackMessage`  to the `PACKAGE_TYPENAME_REGEX` mask which is defined as
https://github.com/ros2/rosbag2/blob/9d64382f833ce211b9841ef8e47047d3d5099d64/rosbag2_storage_mcap/src/message_definition_cache.cpp#L33-L34 
As you can see we have expectation for data types names to be in format `(foo_msgs/Bar or foo_msgs/msg/Bar)`
in case of hidden actions we have `action` instead of `msg` keyword.

emersonknapp, 2023-04-14:
The message definition cache was moved recently into the core, will it now have this problem for all storage implementations?

russkel, 2023-05-04:
I am running into this as well. Except it seems to be intermittent, I am assuming it's timing dependent for some reason?

emersonknapp, 2023-05-19:
> I am running into this as well. Except it seems to be intermittent, I am assuming it's timing dependent for some reason?

I don't think that makes sense. Maybe there's some timing aspect of whether you discover the topic at all? In the cases when you don't get the error, are you recording the topic? By the current code, `action_tutorials_interfaces/action/Fibonacci_FeedbackMessage` will always be invalid, it doesn't match the builtin regex.

russkel, 2023-05-21:
> I don't think that makes sense. Maybe there's some timing aspect of whether you discover the topic at all? In the cases when you don't get the error, are you recording the topic? By the current [...]

I don't really know except sometimes ros2bag crashes, sometimes it didn't. Bit confusing. Maybe I'll have to look into it more.

emersonknapp, 2023-06-13:
I'm going to close this as fixed in #1350 - (iron backport #1352), which eliminates the crash for unknown schemas. If we want to track "get and store schema of implicit action submessages" then I will suggest we file a followup ticket for it.

As always, if I've missed something please feel free to reopen with the details.

closed 2023-06-13 by emersonknapp

msmcconnell, 2023-07-13:
Is this fix going to be backported to humble?

emersonknapp, 2023-07-17:
I believe it was in https://github.com/ros2/rosbag2/pull/1398 but perhaps there hasn't been a new humble release yet

msmcconnell, 2023-07-17:
@emersonknapp Awesome thanks! Based on that date my build might just be old. I'll try again after the next sync.