ros2/rosbag2 #1302: Rosbag record with start-paused feature ingores `/tf_static`

michalpelka, 2023-04-20:
## Description
We are recording the rosbag with such parameters:
```
ros2 bag record  /odom /tf /tf_static   --use-sim-time --start-paused --max-cache-size 10737418240 --no-discovery
```
Our system produces a simulated clock. We need to start in a "paused" state to mitigate the issue raised by my colleague #1276.
The `ros2 bag record` does not record the one message published in `\tf_static`.
 If you open created database and run such query, it will return 0:
 ```
 SELECT COUNT (messages.data) FROM messages INNER JOIN topics ON messages.topic_id== topics.id WHERE topics.name LIKE "/tf_static" 
 ```
 
### Expected Behavior
I would like to `ros2 bag record`  to keep messages that are latched (like tf_static) and record it even with `--start-paused`. Please adjust the correct timestamp to not bring behavior from #1276.

### Actual Behavior
There are no recorded messages in '/tf_static' topic.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Open one terminal and run `ros2 run tf2_ros static_transform_publisher 1 2 3 0.5 0.1 -1.0 foo bar`
2. Open the second terminal and run `ros2 bag record --start-paused /tf_static`
3. Hit space to resume and then CRTL+C to stop recording
4. Investigate created bag with `ros2 bag info`, it will have 0 messages recorded 

Note : if you omit ` --start-paused` in point 2, the resulting bag will  contain expected one message.

## System (please complete the following information)
 - OS:Ubuntu Bionic
 - ROS 2 Distro: Humble
 - Version: release
 
## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2023-04-20:
@michalpelka I have a suspicious that `--no-discovery` is the key in your scenario with `--start-paused` flag.
Can you please try it without `--no-discovery` flag?

MichaelOrlov, 2023-04-20:
@michalpelka Sorry for confusion from my previous message this is certainly unrelated to the `--no-discovery` flag.
`--start-paused` behave as expected and probably not what you need for your scenario.
In paused mode recorder receiving messages and dropping them without saving to the storage. 
It was made by design and we can't do better since if subscription already created callback will be called on message arrive. We can't pause on subscription level.
https://github.com/ros2/rosbag2/blob/18216763ac13d942ee582c4b658962e49179ba3d/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L431-L444

emersonknapp, 2023-05-26:
Agreed that this seems like a bad interaction between latched topics (transient local durability) and `--start-paused`. Maybe there is some fix around waiting to create subscriptions until first `resume`, at least for topics with transient local durability? In any case, that's a little finicky and will take some care.

For this particular case, I think the real fix is https://github.com/ros2/rosbag2/pull/1354. That will backport easily into Iron and should require only a little tweaking to backport to Humble, there are no required API changes, just documentation and implementation.

MichaelOrlov, 2024-01-30:
- Closing as the original issue described in #1276 has been fixed in #1378 and backported to iron in #1391.
- As was mentioned in the analysis here https://github.com/ros2/rosbag2/issues/1302#issuecomment-1516990064 the `--start-paused` feature works as designed and there are no bugs there.

closed 2024-01-30 by MichaelOrlov