ros2/rosbag2 #1325: build error

goooodday, 2023-05-06:
## Description
build error, 

## System (please complete the following information)
 - OS: MacOS Ventura 13.3.19(a)
 - ROS 2 Distro: Humble
 - Version: main
 - H/W: MacBook Pro 16-inch, 2019
2.3GHz 8core Intel Core i9

## Additional context

bellow, command & error message

<pre>
$cd ~/ros2_humble/                                                                              
colcon build --symlink-install --packages-skip-by-dep python_qt_binding
</pre>

<< error message >>

<pre>
Starting >>> rosbag2_test_common
Starting >>> shared_queues_vendor
Starting >>> zstd_vendor
Starting >>> sqlite3_vendor
Starting >>> rosbag2_interfaces
Starting >>> rosbag2_performance_benchmarking_msgs
Starting >>> rosbag2_test_msgdefs
--- stderr: zstd_vendor
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< zstd_vendor [1.57s, exited with code 1]
--- stderr: shared_queues_vendor
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< shared_queues_vendor [1.60s, exited with code 1]
Aborted  <<< rosbag2_test_msgdefs [1.51s]
Aborted  <<< rosbag2_test_common [1.64s]
Aborted  <<< rosbag2_interfaces [1.55s]
Aborted  <<< sqlite3_vendor [1.56s]
Aborted  <<< rosbag2_performance_benchmarking_msgs [1.54s]

Summary: 0 packages finished [2.25s]
  2 packages failed: shared_queues_vendor zstd_vendor
  5 packages aborted: rosbag2_interfaces rosbag2_performance_benchmarking_msgs rosbag2_test_common rosbag2_test_msgdefs sqlite3_vendor
  6 packages had stderr output: rosbag2_interfaces rosbag2_test_common rosbag2_test_msgdefs shared_queues_vendor sqlite3_vendor zstd_vendor
  16 packages not processed
</pre>

MichaelOrlov, 2023-05-06:
@goooodday You got failure in vendor packages. Most probably due to the absent access to the corresponding git repositories during the build.  
Please double check that you have access to the `https://github.com/facebook/zstd.git` during the build process.

emersonknapp, 2023-05-25:
Were you able to successfully build any packages? This is a basic error - "cannot find ament_cmake", meaning that the core installation is not present in your current path environment.

* What's in your workspace? Just rosbag2, or all of ros2?
* If just rosbag2, did you `source $HUMBLE_INSTALL_PATH/setup.bash` first?

MichaelOrlov, 2023-06-03:
- Closing as a stale issue without a response from the author.

closed 2023-06-03 by MichaelOrlov

MichaelFishmanBDAII, 2023-09-08:
I had the same error. The following steps worked for me:

Follow this [guide](https://docs.ros.org/en/humble/How-To-Guides/Setup-ROS-2-with-VSCode-and-Docker-Container.html) to get `ros2` working in a `devcontainer` in `VSCode`

Instead of `mkdir src`, clone `rosbag2` repo into `src`, then put the `.devcontainer` folder in there.
The directory structure will look like

```
ws_[project]
├── cache
|   ├── [ROS2_DISTRO]
|   |   ├── build
|   |   ├── install
|   |   └── log
|   └── ...
|
├── src
    ├── .devcontainer
    │   ├── devcontainer.json
    │   └── Dockerfile
    ├── README.md
    └── <all the other files from the root of the`rosbag2` repo>
```

Then run these commands

```sh
source /opt/ros/$ROS_DISTRO/setup.bash
# From `ws` (NOT ws/src)
colcon build --packages-up-to rosbag2
```
At the time of this comment, this works for the `humble` branch of `rosbag2`. The `rolling` branch gives me an error described in [this PR](https://github.com/ros2/rosbag2/pull/1454).