ros2/rosbag2 #1326: Windows Signal Handling

mjcarroll, 2023-05-09:
I have spent some time investigating the cause of test failures on Windows due to unclean shutdown issues.  Here are some notes that I captured so that I can come back and look at this after the Iron release (or so someone else can pick it up).

I started with a candidate disabled test: https://github.com/ros2/rosbag2/blob/rolling/rosbag2_tests/test/rosbag2_tests/test_rosbag2_record_end_to_end.cpp#L214-L272

This test does a simple recording and verifies that the command line tooling shuts down cleanly and writes all necessary data on the way down.  Currently, on Windows, this test fails.

The recording process is started via the `CreateProcess` call on Windows [(rosbag2 code)](https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/rosbag2_test_common/include/rosbag2_test_common/process_execution_helpers_windows.hpp#L40-L53) [(msdn ref)](https://learn.microsoft.com/en-us/windows/win32/api/processthreadsapi/nf-processthreadsapi-createprocessa).

The recording process is stopped via `GenerateConsoleCtrlEvent` [(rosbag2 code)](https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/rosbag2_test_common/include/rosbag2_test_common/process_execution_helpers_windows.hpp#L108-L120) [(msdn ref)](https://learn.microsoft.com/en-us/windows/console/generateconsolectrlevent).

* GenerateConsoleCtrlEvent fires either a `CTRL_C_EVENT` (equivalent to sigint) or a `CTRL_BREAK_EVENT` (closer to sigterm).
  * We must also specify the process group to send the event to, otherwise it is sent to everything attached to the console (including the ros nodes used by the test).
  * The only way to get the process group is to set the  `CREATE_NEW_PROCESS_GROUP` flag when we do `CreateProcess` and store it somewhere.
  * When a new process group is created, the Ctrl-C event handler is *disabled* by default in the new process:
   > When a process is created with CREATE_NEW_PROCESS_GROUP specified, an implicit call to SetConsoleCtrlHandler(NULL,TRUE) is made on behalf of the new process; this means that the new process has CTRL+C disabled. This lets shells handle CTRL+C themselves, and selectively pass that signal on to sub-processes. CTRL+BREAK is not disabled, and may be used to interrupt the process/process group.

  * Interestingly, using `std::signal` does _not_ automatically re-enable the sigint handler, you must explicitly handle the windows signals.

The best approach would be to add a parallel set of handlers in `rclcpp::SignalHandlers` and potentially in `test_rosbag2` to make sure we are explicitly handling windows signals, and I believe that will clear up some of these shutdown issues.  There is an example on how to do this here (https://learn.microsoft.com/en-us/windows/console/registering-a-control-handler-function)

MichaelOrlov, 2023-05-20:
@mjcarroll I am assigning this task to myself. I have good progress in sending and catching the CTRL+C signal on Windows.
Hopefully will be able to make a fix in rosbag2 without breaking API/ABI.

MichaelOrlov, 2023-05-20:
- Relates #1270

closed 2026-10-29 by clalancette