ros2/rosbag2 #1327: Bag record help message errors and inconsistencies

hwoithe, 2023-05-10:
## Description
There are few errors and inconsistencies in the messages produced by `ros2 bag record --help`.

- Missing period at the end of the sentence. The backslash also seems out of place compared to how the other help messages are constructed.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L57

- Missing period at the end of the sentence.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L61

- Missing period at the end of the sentence.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L107

- Missing space at the end of the string.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L121

- Missing space at the end of the string.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L123

- Missing space at the end of the string. Change magitude to magnitude.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L124

- Missing space at the end of the string.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L125

- Missing space at the end of the string.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L138

- Missing period at the end of the sentence.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L162

- Missing period at the end of the sentence.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/verb/record.py#L167

- Is the "R|" a mistake?
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/api/__init__.py#L199

- Change overriden to overridden.
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/ros2bag/ros2bag/api/__init__.py#L200

### Expected Behavior
Consistent formatting and no errors in the help message.

### Actual Behavior
There are a few errors and inconsistencies in the help message.

## To Reproduce
Execute `ros2 bag record --help`.

## System
 - OS:  Ubuntu Jammy
 - ROS 2 Distro: Iron
 - Version: prerelease

clalancette, 2023-05-11:
See #1329 for fixes.

hwoithe, 2023-05-11:
Thank you for the quick turnaround.

PR #1329 (2023-05-12): Cleanup the help text for ros2 bag record.
Add periods where needed, reindent the code as necessary, and add spaces where needed.

This will fix #1327
closed 2023-05-12 by clalancette