ros2/rosbag2 #1332: ros2 bag record: Implausible warnings

ahemwe, 2023-05-15:
## Description
`ros2 bag record `shows implausible and non-reproducible warnings while not recoding some topics even with `-a ` option.

### Expected Behavior
Full recording, no warning.

### Actual Behavior
Warnings on arbitrary topics indicating non-compatible reliabiity settings between different publisher.
But there is actually only a single publisher.

## To Reproduce

### Setup publishing node with sensor qos profile

```
  // Quality of service: Sensor default, but queue size of 1
  static const rclcpp::QoS default_qos = rclcpp::QoS(rclcpp::KeepLast(1), rmw_qos_profile_sensor_data);

  // Publisher
  pub = this->create_publisher<my_pkg::msg::MyMsgT>("myMsg", default_qos);
```
### Confirm settings for this topic
``` bash
$ros2 topic info /myMsg --verbose
Type: my_pkg/msg/MyMsgT

Publisher count: 1

Node name: _NODE_NAME_UNKNOWN_
Node namespace: _NODE_NAMESPACE_UNKNOWN_
Topic type: my_pkg/msg/MyMsgT
Endpoint type: PUBLISHER
GID: 01.0f.42.19.bd.1a.17.df.01.00.00.00.00.00.19.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Subscription count: 0
```

### Run record

This command has to be repeated several times in an setup with multiple published topics. The warning pops up on arbitrary topics.

``` bash
$ros2 bag record -a
[INFO] [1683889007.313020116] [rosbag2_storage]: Opened database 'rosbag2_2023_05_12-12_56_47/rosbag2_2023_05_12-12_56_47_0.db3' for READ_WRITE.
[INFO] [1683889007.323326234] [rosbag2_transport]: Listening for topics...
....
[WARN] [1683889007.443336458] [rosbag2_transport]: A new publisher for subscribed topic /myMsgwas found offering RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT, but rosbag already subscribed requesting RMW_QOS_POLICY_RELIABILITY_RELIABLE. Messages from this new publisher will not be recorded.
```

## System (please complete the following information)
 - OS: Ubuntu 20.04, either on arm64 or on WSL2
 - ROS 2 Distro: Foxy
 - Version: ros-foxy-ros2action/focal,now 0.9.13-1focal.20230317.044239 arm64

## Additional context

This is a cross-post with [answers.ros.org]( https://answers.ros.org/question/415280/ros2-bag-record-implausible-warnings/) where I didn't get any answer.

There is the same (?) question mentioned for Galactic in #830 but without any result for foxy.

MichaelOrlov, 2023-05-16:
Created  backporting PR https://github.com/ros2/rosbag2/pull/1335

MichaelOrlov, 2023-05-18:
- Closing as fixed in #1335

closed 2023-05-18 by MichaelOrlov