ros2/rosbag2 #1336: Broken regex check for SequentialWriter after MCAP merge

rshanor, 2023-05-17:
## Description
I am using `rosbag2_py.SequentialWriter` class to record service requests and responses to a bag file. Code that previously worked is now throwing a `ValueError: Invalid topic_type: pylon_ros2_camera_interfaces/srv/SetGamma_Request`. It looks like https://github.com/ros2/rosbag2/pull/1163 added some regex checking to the message type which is not valid for service request and response messages: https://github.com/ros2/rosbag2/blob/rolling/rosbag2_cpp/src/rosbag2_cpp/message_definitions/local_message_definition_source.cpp#L33

```
>       self.writer.create_topic(topic_info)
E       ValueError: Invalid topic_type: pylon_ros2_camera_interfaces/srv/SetGamma_Request
```

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Rolling
 - Version: latest

MichaelOrlov, 2023-05-18:
- Relates to the https://github.com/ros2/rosbag2/issues/1286

MichaelOrlov, 2023-05-18:
@rshanor Could you please clarify what is specifically in the service request/responses causing this failure?

rshanor, 2023-05-18:
@MichaelOrlov here is a minimal way to repro.

```py
import rosbag2_py
from std_srvs.srv import SetBool
import tempfile
import uuid

writer = rosbag2_py.SequentialWriter()
storage_options = rosbag2_py._storage.StorageOptions(
    uri=f"/tmp/{uuid.uuid4()}", storage_id="sqlite3"
)
converter_options = rosbag2_py._storage.ConverterOptions("", "")
writer.open(storage_options, converter_options)

msg = SetBool.Request()
msg_type = type(msg)
msg_module = msg_type.__module__
msg_name = msg_type.__name__
msg_mod_split = msg_module.split(".")
topic_type = f"{msg_mod_split[0]}/{msg_mod_split[1]}/{msg_name}"
full_name = "/example"
topic_info = rosbag2_py._storage.TopicMetadata(
    name=full_name,
    type=topic_type,
    serialization_format="cdr",
)
writer.create_topic(topic_info)
```

MichaelOrlov, 2023-05-18:
@emersonknapp @james-rms If you have time, your help with this issue and relevant https://github.com/ros2/rosbag2/issues/1286 will be very appreciated.

emersonknapp, 2023-05-19:
Assigning to myself and looking into it now. This one should be an easy fix, the message definition scraper had a hardcoded `/msg/` infix for types. For services, the `.msg` for `_Request` and `_Repsonse` always exist in the install tree, so the data exists.

For #1286 there is the same problem of "can't find the definition" but it's not a complete duplicate, that one deals with Actions which do not have all subtypes as plain files in the install tree.

emersonknapp, 2023-06-13:
The crash scenario was fixed in #1350 (iron backport #1352) - but I'm going to leave this open until #1341 is finished, as that will extend the full feature to services, which is nice to have.

MichaelOrlov, 2024-01-30:
- Closing this issue as fixed
1. The crash was fixed in the #1350. 
2. The follow-up PR #1341 was closed as it wouldn't be done. At least for while.
3. Now we have the special feature and CLI parameters for recording services and no need to workaround it and writing own services recorder for that.

closed 2024-01-30 by MichaelOrlov