ros2/rosbag2 #1343: The description of `--ignore-leaf-topics` doesn't match the implementation

Barry-Xu-2018, 2023-05-22:
## Description

The description of `--ignore-leaf-topics` doesn't much the implementation

```
--ignore-leaf-topics  Ignore topics without a publisher.
```

The implementation of check function is 
https://github.com/ros2/rosbag2/blob/65c889e1fa55dd85a148b27b8c27dadc73238e67/rosbag2_transport/src/rosbag2_transport/topic_filter.cpp#L81-L87

So I think the description should be `Ignore topics without a subscription`

## System (please complete the following information)
 - OS: Ubuntu Bionic
 - ROS 2 Distro: rolling
 - Version: rolling branch

Barry-Xu-2018, 2023-05-23:
> Should be Jammy? since the issue is detected with rolling?

Yes. OS is Jammy. I will fix it.  
Thank you for your reminder !

Barry-Xu-2018, 2023-05-23:
I close this issue, since the fix #1344 has already been merged.

closed 2023-05-23 by Barry-Xu-2018

MichaelOrlov, 2023-05-23:
- Closing as resolved in #1344