ros2/rosbag2 #1348: Some chunks appears to be corrupted after recording

dimaxano, 2023-05-24:
## Description
rosbag recording is started from the launch file with the command: 
```python
ExecuteProcess (
            condition = IfCondition(use_data_recording),
            cmd = [
                "ros2", "bag", "record",           
                "--regex=.+",
                "--exclude="
                    "/camera(.+)|"
                    ".+/voxel_grid|"
                    ".+/processed_pointcloud|"
                    "/cloud|"
                    "/raw_data|"
                    "/extended_scan|"
                    "/global_costmap(.+)",
                f"--qos-profile-overrides-path={qos_profile_overrides_file}",
                "--max-bag-size=3221225472",    # 3Gb of raw data (~11x less after compression)
                "--max-bag-duration=1800",      # seconds
                "--compression-mode=file",
                "--compression-format=zstd",
                "--compression-threads=4",
                "--include-hidden-topics",
            ],
            log_cmd = False,
            cwd     = rosbags_storage_path,
            name    = 'rosbag_record',
            output  = 'screen'
        )
```
After the recording is finished some rosbags chunks are impossible to uncompress. `ros2 bag play` says `Could not open './rosbag2_2023_05_22-05_59_41_1.db3' with 'sqlite3'. Error: Failed to setup storage. Error: Error when processing SQL statement. SQLite error (11): database disk image is malformed"

### Expected Behavior
No malformed chunks are present.

### Actual Behavior
There are two types of corrupted chunks:
1. Recognized as an zstd archive, but impossible to decompress. The output of `zstd -d /path/to/chunk`: Read error (39) : premature end
2. Doesn't even recognize as zstd archive. The output of `zstd -d /path/to/chunk`: unsupported format 

## To Reproduce
Haven't found a way to reproduce that

## System (please complete the following information)
 - OS: Ubuntu 22.04 (inside docker)
 - ROS 2 Distro: Humble
 - Version: [release, branch, or commit hash]

## Additional context
Here is the [link](https://drive.google.com/file/d/1mjfDA8qTsNttSsS41o8dAFWL0CAfx2nY/view?usp=share_link) to such rosbag.

Chunks 1,3 are the first type of corrupted chunks and chunks 5,7 are the second type of corrupted chunks.

I check the disk where that recording happened with `badblocks` tool and no errors were found on the disk.

emersonknapp, 2023-05-25:
A few questions to help understand the issue:

1. How was recording terminated? Ctrl+C on the `launch` command terminal?
1. You've only experienced this once? Or was it several times? 
1. Was the `metadata.yaml` file present when it happened? I'm wondering if it has something to do  with a bad shutdown case

dimaxano, 2023-05-26:
I'm unsure if that answers your concern about the shutdown, but problematic chunks are never the last. They are always somewhere in the middle of the bag.

1. As the recording is running in the docker, we call `docker stop` (which sends SIGTERM until a timeout and then SIGKILL)
2. I see such chunks quite often, I guess every 4-6 hours of recording (with the given above params for chunks split criteria)
3. Corrupted chunks are also present in the bags with `metadata.yaml` generated by `ros2 bag record`

emersonknapp, 2023-05-26:
Thanks for the info - can you tell me anything else about your platform, such as how much memory there is?

It's hard to tell what's going on here, but I am currently wondering if there is some race condition around https://github.com/ros2/rosbag2/blob/rolling/rosbag2_compression/src/rosbag2_compression/sequential_compression_writer.cpp#L300 where the compression thread works on the file while the sqlite3 process still tries to write at it, corrupting the archive.

dimaxano, 2023-05-30:
Thanks for taking a look at that, @MichaelOrlov 

As I said earlier corrupted chunks are never the last ones i.e among 30 chunks # 5, 15 maybe corrupted and the rest are fine. As I understand, by the time `SIGTERM` / `SIGKILL` was received, the recorder already done with the corrupted chunks.

Do you think it makes sense?

dimaxano, 2023-06-02:
friendly ping @MichaelOrlov

I can use built from source rosbag2 and bring there some debug messages to help with identifying where it fails 
If that can help

MichaelOrlov, 2023-06-03:
@dimaxano While I need some time to verify the logic on the Humble branch (it might be different from Rolling), could you please try to reproduce this issue with MCAP storage format? i.e. using `-s mcap` option during recording.

MichaelOrlov, 2023-06-03:
@dimaxano We have by default a number of compressor workers threads equal to the `std::thread::hardware_concurrency()`.
Therefore possible situation that we are starting to compress a new bag in parallel while the previous one hasn't been finished yet.
Please consider a possible scenario that could be that at the moment when you are sending `SIGTERM` the signal compressor already managed to compress the latest bags and is a bit stuck with the compression of some bag which is supposed to be early. 
Since we have multiple threads which could run compression for multiple files in parallel in the background and there is no guarantee that job threads will finish compression in order its start. It could be a  situation the middle bags haven't been compressed yet at the moment when `SIGTERM / SIGKILL` arrived.

dimaxano, 2023-06-06:
@MichaelOrlov I'll take an MCAP test next week

I'll also try to reproduce the problem by sending only SIGTERM, without SIGKILL after the timeout. Will see If I succeed

emersonknapp, 2023-06-06:
Oh - I just realized there is a bit of debug log printing there when a file finishes compressing, but it is at DEBUG level and I don't actually know if there is a good way for us to set the logger level from the commandline (https://github.com/ros2/rosbag2/issues/288). This would be very possible if the recorder were an exposed node such that we could pass `--ros-args` to it in the `ros2 run` CLI - but this also hasn't been worked on yet (https://github.com/ros2/rosbag2/issues/902)

dimaxano, 2023-09-04:
Hi!

I finally tested with `mcap` and I'm not able to reproduce the issue anymore. Does this explain anything about the original problem?

Anyway, my issue seems solved. Thank you for the help!

closed 2023-09-04 by dimaxano

MichaelOrlov, 2023-09-06:
@dimaxano MCAP storage plugin uses less CPU and memory resources. It might be that with MCAP your system has more resources for zstd file compressor threads and they are finishing in time before the shutdown happened.