ros2/rosbag2 #1351: Failed to build from source [-Werror: Warning treated as error]

xkaraman, 2023-05-25:
## Description
I am trying to build the ROS2 from source, but it fails on mcap_vendor packages with the error
```
~/ros2_workspaces/ros2_iron_ws/build/mcap_vendor/_deps/mcap-src/cpp/mcap/include/mcap/writer.inl:745:46: error: unused variable ‘channelId’ [-Werror=unused-variable]
     for (const auto& [channelId, messageIndex] : currentMessageIndex_) {
                                              ^
cc1plus: all warnings being treated as errors

```

## FIx
1. Probably this should be resolved from upstream package at foxglove if possible.
or
2. Removing the -Werror in CMakelists. ( This builds succesfully in my OS)

### Expected Behavior
I expected the build with the commnad `colcon build --packages-up-to mcap_vendor` to complete succcesfully

### Actual Behavior
Building failing

## To Reproduce
1. Follow ROS2 Official Build from source instuctions
4. Modify the rosdep install command to `rosdep install --from-paths src --ignore-src -y --rosdistro iron --skip-keys  "fastcdr rti-connext-dds-6.0.1 urdfdom_headers ignition-cmake2 ignition-math6 python3-pykdl"` due to incompatibilities from my OS (Ubuntu 18.04: bionic)
5. `colcon build --packages-up-to mcap_vendor` 
6. Logs print error '...' **

## System (please complete the following information)
 - OS:Ubuntu Bionic 18.04
 - ROS 2 Distro: Iron
 - Version: latest Iron

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2023-05-25:
@xkaraman What version of compile do you use?
We don't see this issue on CI nor on our local setup.
cc: @james-rms

emersonknapp, 2023-05-25:
Note: Ubuntu 18.04 Bionic is not a supported platform for ROS 2 Iron. See https://www.ros.org/reps/rep-2000.html#iron-irwini-may-2023-november-2024 - only Ubuntu 22.04  Jammy is officially supported. That said, this shouldn't be an issue given that the line in question is very generic code.

I'm trying it out for the sake of curiosity, and run into a number of issues:

1. I tried following the official setup instructions at https://docs.ros.org/en/iron/Releases/Release-Iron-Irwini.html - and already this doesn't work on Bionic, there is no package `ros-dev-tools` in APT on Bionic to install

1. Bionic comes with CMake 3.10 and the `ament_cmake` package for Iron requires >= 3.12, so you must have changed something there to get as  far as the `mcap_vendor` build

So - I install CMake 3.25 from the [Kitware APT repo](https://apt.kitware.com) and move on

2. `ament_cmake_core` uses Python's `importlib.resources`. However, this is new in Python 3.7 and Bionic only ships with Python 3.6.9, so you must have installed a newer Python version somehow

At this point I'm giving up, I think you've misrepresented your platform somehow - if you're getting the results that you've posted here, you have heavily modified your system somehow in a way that I am not able to reverse engineer from the information presented.

Please provide more information if you still need help - but I do recommend using a supported platform if you wish to use ROS 2  Iron, there are no guarantees that it will work anywhere else. I'm going to close this issue as `wontfix` for now, but please reopen if you have more information to make the issue reproducible.

closed 2023-05-25 by emersonknapp

xkaraman, 2023-05-26:
Hey there,
Thanks for the quick response. I am fully aware that Iron release is not supported on Ubuntu 18.04. i was just curious as you were, since i wanted to experiment on this. Indeed, my setup is heavily modified, so i it can be reasonable to fail to build.

For the sake of curiosity though, i did the install procedure on clean docker containers: one for `ubuntu:bionic` and one for `ubuntu:jammy`. Apparently, this of course doesn't happen on the jammy as your CI also suggests, but this remains a question for me. 

Since the `mcap` src fetched from `mcap_vendor` package is the same exact version and hash and `mcap_vendor` `CMakeLists.txt` both add the `-Werror` compile option, isn't the expected behaviour to fail the build? Issue #720 and  PR #722  also seems to suggest to remove `-Werror` from every package involved in `rosbag2` but yet in `mcap_vendor` still remains.

I don't want to waste any of your time here, just trying to understand why jammy builds successfully independent from bionic build.

Thanks again for the info and try so far.

You can find the commands used in docker containers and some notes regarding your points. 

1. `ros-dev-tools` is just a convenience package that contain multiple `python3-packages, git and cmake` as noted [in this ros discourse post](https://discourse.ros.org/t/ros-developer-tools-now-in-binary-form/29802)
2.  Indeed CMake on bionic is versions behind, also followed the same to get to the latest
3.  Python3 on Bionic is indeed 3.6.9 and that's the only version on docker. Nowhere it fails about `importlib.resources` as one might expect because it get's installed through `rosdep` and `pip` (probably), as you can see when issuing `pip3 list` in bionic container.

<details>
  <summary>Bionic Docker</summary>

```
apt update && apt install locales
locale-gen en_US en_US.UTF-8
update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
export LANG=en_US.UTF-8

locale

apt install software-properties-common
add-apt-repository universe

# Install Cmake Latest Version
apt install wget
wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | tee /etc/apt/trusted.gpg.d/kitware.gpg >/dev/null
apt-add-repository "deb https://apt.kitware.com/ubuntu/ $(lsb_release -cs) main"
apt install kitware-archive-keyring
rm /etc/apt/trusted.gpg.d/kitware.gpg
apt install cmake

apt install python3
apt install curl

# Add ROS 2 Repo
curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

# https://colcon.readthedocs.io/en/released/user/installation.html
# Add Install colcon
sh -c 'echo "deb [arch=amd64,arm64] http://repo.ros2.org/ubuntu/main `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'
curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -
apt update
apt upgrade
apt install python3-colcon-common-extensions

## Install required deps
apt install python3-vcstool
apt install python3-rosdep
apt install python3-pip

mkdir -p ~/ros2_iron/src
cd ~/ros2_iron
vcs import --input https://raw.githubusercontent.com/ros2/ros2/iron/ros2.repos src

rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers python3-pykdl ignition-math6 ignition-cmake2"

cd ~/ros2_iron/
colcon build --packages-up-to mcap_vendor
```
</details>

<details>
  <summary>Jammy Docker</summary>

```
apt update && apt install locales
locale-gen en_US en_US.UTF-8
update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
export LANG=en_US.UTF-8

locale

apt install software-properties-common
add-apt-repository universe

apt install curl

# Add ROS 2 Repo
curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

apt install -y \
  python3-flake8-docstrings \
  python3-pip \
  python3-pytest-cov \
  python3-flake8-blind-except \
  python3-flake8-builtins \
  python3-flake8-class-newline \
  python3-flake8-comprehensions \
  python3-flake8-deprecated \
  python3-flake8-import-order \
  python3-flake8-quotes \
  python3-pytest-repeat \
  python3-pytest-rerunfailures \
  ros-dev-tools
  
mkdir -p ~/ros2_iron/src
cd ~/ros2_iron
vcs import --input https://raw.githubusercontent.com/ros2/ros2/iron/ros2.repos src

apt upgrade
  
rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers python3-pykdl ignition-math6 ignition-cmake2"
```
</details>

emersonknapp, 2023-05-26:
Pretty clear from all that, actually:

``` C++
// lambda_capture.cpp
#include <cstdio>
#include <unordered_map>

int main()
{
  const std::unordered_map<int, int> currentMessageIndex_ = {
    {1, 2},
    {3, 4}
  };
  for (const auto & [channelId, messageIndex] : currentMessageIndex_) {
    printf("%d\n", messageIndex);
  }
}
```

``` bash
$ g++-8 -std=c++1z -Wall -Wextra -Wpedantic lambda_capture.cpp

$ g++-7 -std=c++1z -Wall -Wextra -Wpedantic lambda_capture.cpp
lambda_capture.cpp: In function ‘int main()’:
lambda_capture.cpp:10:45: warning: unused variable ‘channelId’ [-Wunused-variable]
   for (const auto & [channelId, messageIndex] : currentMessageIndex_) {
```

GCC 8 doesn't think that  it's a problem, but GCC 7 does. I'll also note I just tried it with clang6 and clang9, which I happened to have installed, and neither of them raised the warning either.

emersonknapp, 2023-05-26:
I think, based on that, my suggestion is to use GCC >= 8 when trying to build Iron from source. The version that actually ships on the supported platform Jammy is gcc11, but that isn't available out of the box on Bionic so you can probably get away with 8

xkaraman, 2023-05-29:
Yes, good to know about the tests you made. I will try to continue the build, but obviously other problems are rising, with version conflicts and more.

Should this bug (guessing it should be) be reported to GCC? or is it the expected behavior because of the "unpacking" of variables?

Again thanks for the help and effort to make me understand what's going on!