ros2/rosbag2 #1365: New release for Humble?

haudren-woven, 2023-06-02:
## Description

I am trying to use rosbag2 in Humble, but this issue: https://github.com/ros2/rosbag2/pull/1225 makes it very annoying to use as my console is either filled with messages from the binary version, or I need to build rosbag2 from source.

## Related Issues

https://github.com/ros2/rosbag2/issues/1139

## Completion Criteria

Release rosbag2 0.15.6 or so for Humble. As far as I can see, a few extra small issues have been fixed, but I don't think it's anything major?

## Implementation Notes / Suggestions

## Testing Notes / Suggestions

emersonknapp, 2023-06-07:
I appreciate the request, but I am wondering if this issue tracker is the right place to track release schedule. It's more of a housekeeping task than a feature request or bug ticket. Though, I don't have a specific idea off the top of my head right now for doing it better - just putting thoughts out

haudren-woven, 2023-06-08:
I agree with it :thinking: Is there any other place where we can have that discussion? Maybe discourse? In any case, this was solved by #1372 , so let me close this :)

closed 2023-06-08 by haudren-woven