ros2/rosbag2 #1366: Python typesys msg type causes assertion errors in other modules

broughtong, 2023-06-02:
## Description
Im trying to read pointclouds from a bag in Rolling, which are stored as `sensor_msgs/msg/PointCloud2`.
When I deserialise these, they become of type `<class 'rosbags.typesys.types.sensor_msgs__msg__PointCloud2'>`.

This causes assertion errors in for example `sensor_msgs_py` which makes a call to `assert isinstance(cloud, PointCloud2)`.
This is failing as the message is not of the correct type `sensor_msgs/msg/PointCloud2`.

Perhaps there is some way to alter the class, I tried [this](https://stackoverflow.com/questions/73420147/how-to-read-custom-message-type-using-ros2bag), but ended up with:
`TypeError: __class__ assignment: 'PointCloud2' object layout differs from 'sensor_msgs__msg__PointCloud2'`

Also tried to register the original type but I didn't get this working.

### Expected Behavior
I can read in a pointcloud from a rosbag and it behaves like a normal pointcloud message in other modules.

### Actual Behavior
Assertion error in sensor_msgs_py: `assert isinstance(cloud, PointCloud2), \
AssertionError: Cloud is not a sensor_msgs.msg.PointCloud2`

## To Reproduce
```
from rosbags.rosbag2 import Reader
from rosbags.serde import deserialize_cdr
from sensor_msgs_py import point_cloud2 as pc

filename = 'somefilename'

with Reader(filename) as reader:

    for connection, timestamp, rawdata in reader.messages():

        if connection.topic == "/ouster/points": #sensor_msgs/msg/PointCloud2
            msg = deserialize_cdr(rawdata, connection.msgtype)
            msg = pc.read_points_numpy(msg) #error occurs here
```

## System (please complete the following information)
Ubuntu Bionic, Rolling

## Additional context

broughtong, 2023-06-03:
Similarly, taking a message from a rosbag and simply republishing it:
`python: /home/george/common_ws/build/std_msgs/rosidl_generator_py/std_msgs/msg/_header_s.c:57: std_msgs__msg__header__convert_from_py: Assertion strncmp("std_msgs.msg._header.Header", full_classname_dest, 27) == 0' failed.
Aborted (core dumped)`

Where the header now has a new class type `<class 'rosbags.typesys.types.std_msgs__msg__Header'>`

emersonknapp, 2023-06-07:
Can you clarify more about the bag you're trying to read? Is this a ROS 1 bag that you're trying playing back in ROS 2 using the `ros1_bridge` installation on Rolling (on Ubuntu Bionic?)

tom-bu, 2024-03-23:
You'll need to convert the msg to sensor_msgs.msg.PointCloud2() messge. Your original msg has all of the data fields, so you can instantiate a PointCloud2 with those fields.

MichaelOrlov, 2024-05-05:
- Closing as a stale issue since there is no response from the author.

closed 2024-05-05 by MichaelOrlov