ros2/rosbag2 #1397: Data in subscription queue lost when terminating record

ahemwe, 2023-06-13:
## Description
I'm running a ros 2 bag record with qos override with 
```
  reliability: best_effort
  history: keep_all
```
This record is stopped by means of sigint.

### Expected Behavior
All data, even the data in queue is written to bag file.
1. Terminate subscription so that queue does not get any new data
2. write all data out of queue to file

### Actual Behavior
Topics with high sampling rate and therefore long queue discarded without completely writing the data in queue.

## To Reproduce
1. Run ros package
3. `ros2 bag record -a -o $targetPath$name --qos-profile-overrides-path qos_override.yaml & sleep 1`
4. `sleep 100`
5. `kill -SIGINT $(pgrep -f 'ros2 bag record')`

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: foxy
 - Version: latest from apt

## Additional context
Topics with sampling rate below 100 m/s have a queue, topics with 500 ms sampling rate are written properly.
I connect to 84 topics.
For a measurement time of 350 s, the fast topics end at ~250 s in my rosbag due to discarded queue.

emersonknapp, 2023-06-13:
Thanks for the bug report, this behavior is a legitimate concern. For our information, have you yet tried your workflow on a newer ROS distribution, such as Humble? Unfortunately ROS 2 Foxy is now end-of-life, and so won't be receiving any new code updates.

ahemwe, 2023-06-14:
Thanks for looking into it.

No, I didn't test as my arm device has an ubuntu 20 image only.

Is `sigint` reasonable from your perspective?

I'll try to individually specify topics to be logged instead of `-a` in order to reduce overall workload.

emersonknapp, 2023-06-14:
Yes, `SIGINT` is the best way to terminate recording for a `ros2 bag record` commandline invocation

closed 2023-06-14 by ahemwe