ros2/rosbag2 #1399: foxy-future rosbag2 version does not work properly with sensor_msgs/msg/Image data and cyclone DDS

KarasevPavel, 2023-06-15:
## Description
We use custom camera package for controlling Hikrobot cameras.

Our custom implemended node for cameras works well on 3 cameras simultaneously and we can achieve 30 fps (30 hz) on the each camera.
Number of hz on all 3 cameras, working simultaneously we have measured by `ros2 topic hz`

### Expected Behavior
We wanted to record topics data from cameras with **rosbag2** from the **foxy-future** branch, built in our workspace and expected to have a bag file with 3 recorded topics from three cameras in it with same frequency (30 hz).

### Actual Behavior
Actual frequency of topics in bag file is less. than 6 hz, sometimes less, than 3 hz.

## To Reproduce

### In the first terminal:
get into ROS workspace
```bash
cd my_ros_ws
```
clear all previous builds:
```bash
rm -rf build/ install/ log/
``` 
turn RMW_IMPLEMENTATION by:
```bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```
build workspace with package for cameras and rosbag2 on **foxy-future** branch:
```bash
colcon build
```
source overlay:
```bash
source install/local_setup.bash
``` 
launch a launch file to start camera nodes
```bash
ros2 launch <launch file name>
```

### In the second terminal:
get into ROS workspace
```bash
cd my_ros_ws
```
turn RMW_IMPLEMENTATION by:
```bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```
run command to check topic hz of the first_camera:
```bash
ros2 topic hz /first_camera_topic
```
You will probably see 30 hz for the all time before starting a recording process

### In the third terminal:
get into ROS workspace
```bash
cd my_ros_ws
```
turn RMW_IMPLEMENTATION by:
```bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```
source overlay:
```bash
source install/local_setup.bash
``` 
run ros2 bag record:
```bash
ros2 bag record /first_camera_topic /second_camera_topic /third_camera_topic
```
Check the second terminal and you will see, how hz drops down within time and stabilize on 3 - 6 hz

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Version: foxy-future

## Additional context
Camera topics use **rclcpp::SensorQos (Best effort)**
Camera topics message type: **sensor_msgs/msg/Image**
RMW: **rmw_cyclonedds_cpp**

emersonknapp, 2023-06-15:
Unfortunately Foxy has hit end-of-life and is now frozen. `foxy-future` though is not frozen like the rest of the distro, I have no problem with this branch getting updates post-EOL. But, it's unlikely that any of the maintainers will have any resources to put towards that.  We'd do our best to review any PRs, but I wanted to set expectations that this all you'd be able to expect from us.

KarasevPavel, 2023-06-16:
Hello @emersonknapp! Thank You for your quick reply!

I have localized the problem yesterday.
The problem was in simultaneous recording a bag and listening topic's hz at the same time with the `ros2 topic hz` command. 
Unfortunately I do not understand which side is the problem: rosbag2 or ros2 cli, but I think it is not a big problem for now and I can record bags without checking topic's rate.

I will make a PR, if I will find a bottleneck a bit later

emersonknapp, 2023-06-16:
`ros2 topic hz` just creates a subscription to the topic, so it's also taking the full bandwidth of the topic, which for image data is quite large. Maybe reducing the network traffic by that much is what made it start working. The `hz` thing is a bit unfortunate, it's possible there would be a way to get the topic metadata without transmitting the entire message. But, that's probably a deep question.

If it's OK with you, I'll close this issue for now. If you have a specific concern feel free to reopen or open a new ticket.

closed 2023-06-16 by emersonknapp

MichaelOrlov, 2023-06-16:
@KarasevPavel I have seen a similar problem before. And it was related to the `ros2 topic hz` and executor implementation.
In addition to what @emersonknapp mentioned we also observed that `ros2 topic hz` occupied 100% of the one CPU core.
Due to the suboptimal implementation in the executor code. Usually, rosbag recorder also consume 100% of the one CPU core.
If you are trying to run it all together on the embedded system with other nodes in parallel it might be crucial in regard to the entire system performance.