ros2/rosbag2 #1406: Error playing a recorderd .mcap ros2 bag : Could not open 'rosbag2_xxx.mcap' with 'sqlite3'.

ronyshaji, 2023-06-30:
Hello
I have an issue with playing a ros2 bag file in .mcap format in my ros2 ecosystem. I got the ros2 bag from another system and i am trying to play the bag file. But I am getting the erorr as follows:

```
[ERROR] [1688114346.119363457] [rosbag2_storage]: Could not open 'rosbag2_xxx.mcap' with 'sqlite3'. Error: Failed to setup storage. Error: Error when processing SQL statement. SQLite error (26): file is not a database
[ERROR] [1688114346.119398254] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'.
[ERROR] [1688114346.119484850] [rosbag2_transport]: Could not read metadata for rosbag2_xxx.mcap. Please specify the path to the folder containing an existing 'metadata.yaml' file or provide correct storage id if metadata file doesn't exist (see help).
```

The .yaml file is available along with the ros2 bag. But still the error comes. What may be the issue?

ronyshaji, 2023-06-30:
@MichaelOrlov Do youo have any idea about this? I found you have some issue like this earlier?

emersonknapp, 2023-06-30:
Could you please provide the information requested from the bug report template? https://github.com/ros2/rosbag2/issues/new?assignees=&labels=bug&projects=&template=bug_report.md&title=

I've copy-pasted the template form into the description, if you would kindly fill out the remaining fields it will help to understand the issue

MichaelOrlov, 2023-06-30:
@ronyshaji I agree with @emersonknapp we need more info.
At first glance it looks like you are trying to playback some mcap file on some old ROS 2 distro which doesn't have proper support for it.

ronyshaji, 2023-07-03:
@emersonknapp and @MichaelOrlov The issue has been updated as per the template.

MichaelOrlov, 2023-07-04:
@ronyshaji Yep. My preliminary guess was correct.  
`ROS 2 Foxy` doesn't support MCAP file format and it is already reached its end of life.

closed 2023-07-04 by ronyshaji