ros2/rosbag2 #1407: Setting compression mode when using rosbag2_py.SequentialCompressionWriter

bmdyrdal, 2023-06-30:
Hi folks,

I'm writing a component for recording specific topics, and wanted to use the rosbag2_py library. 
I'm having some difficulties finding out how I can set the compression mode? I see that you can set it in the bagmetadata, but I don't see how I can use that?
Anyone who has any examples of how I can do this? 

Thanks in advance!

MichaelOrlov, 2023-06-30:
@bmdyrdal What storage backend and ROS 2 distro are you going to use?

MichaelOrlov, 2023-06-30:
If this is for Iron or Rolling - we have a special field in the `StorageOptions`
https://github.com/ros2/rosbag2/blob/b6f5b43feb43a6481f4f203ee14ad10f02c7a55d/rosbag2_storage/include/rosbag2_storage/storage_options.hpp#L53
Please refer to the https://github.com/ros2/rosbag2#recording-with-a-storage-configuration and https://github.com/ros2/rosbag2/blob/rolling/rosbag2_storage_mcap/README.md#writer-configuration and https://github.com/ros2/rosbag2/blob/rolling/rosbag2_storage_sqlite3/README.md#storage-configuration-file 
to see how to write config files for storage backend.

simonsag96, 2023-07-11:
As far as I see the constructor of the Sequential Compression Writer has an argument of type `rosbag2_compression::CompressionOptions` 

https://github.com/ros2/rosbag2/blob/d788e518b3a93acb61226d993e3d3095913f16b1/rosbag2_compression/include/rosbag2_compression/sequential_compression_writer.hpp#L61-L63

However, this argument is not bound to Python accordingly in `rosbag2_py`
https://github.com/ros2/rosbag2/blob/d788e518b3a93acb61226d993e3d3095913f16b1/rosbag2_py/src/rosbag2_py/_writer.cpp#L109-L110

Furthermore, the `rosbag2_compression::CompressionOptions` is not bound to Python.

So wouldn't it be correct to adapt the pybind11 code in rosbag2_py to adapt the API of the cpp version?

bmdyrdal, 2023-07-11:
@MichaelOrlov Using Humble, store it using sqlite3 in our file system. 
Works fine using it from the command line.

@simonsag96 Is spot on with his comment! :+1: 

While waiting for this I ended up writing a python node which starts the recording as a subprocess.

MichaelOrlov, 2023-07-13:
@bmdyrdal @simonsag96 Yes. The solution would be to provide pybind11 code in rosbag2_py package to map `rosbag2_compression::CompressionOptions` to Python and use it as the default parameter in the Python version of the `SequentialCompressionWriter`. 

But unfortunately, I don't have time for this task.  If some of you can provide a PR it would be great, otherwise, this task will go in the backlog.

simonsag96, 2023-07-13:
Thanks for your answer @MichaelOrlov!

I could implement this over the course of next week!

Regarding Testing: Do you know if it's possible to test my implementation in combination with my binary install (building the adapted pkgs in an overlay workspace)  or do I have to build the whole ros2 stack to get accurate results?

asymingt, 2023-07-17:
I needed this myself today, so I put together this PR: https://github.com/ros2/rosbag2/pull/1425

closed 2023-07-20 by MichaelOrlov