ros2/rosbag2 #1412: rosdoc2 Documentation (Doxyfile)

emersonknapp, 2023-07-07:
## Description

Create Doxyfiles for the projects in this repo, making at least https://build.ros2.org/view/Rdoc/job/Rdoc__rosbag2__ubuntu_jammy_amd64/ pass for all packages present in this repo, therefore having an entry for all of them in https://docs.ros.org/en/rolling/p/

This is a very basic start, but would get us on the road to better documentation, and a place to put documentation so it's discoverable. Rosbag2 is a very important tool for ROS users, but it's pretty much always been the case that only the contributors really know all the features it has available.

sangteak601, 2024-03-14:
I will take a look at this.

EmmanuelMess, 2025-05-26:
As a note: some documentation for C++ is available, but not for python.