ros2/rosbag2 #1418: "Finalize" bag files when recording

ianstarz, 2023-07-12:
## Description

Provide some sort of signal to other processes when a bag file is ready for downstream processing (like uploading).

## Related Issues

None that I can find easily.

## Completion Criteria

Another process can monitor a directory where bags are recorded to, and understand when a bag file will no longer be modified by any other ROS process.

## Implementation Notes / Suggestions

One way this could work is to mark the files as readonly. This would work across POSIX systems in case that is a constraint.

## Testing Notes / Suggestions

I would need help with this. Not sure how you currently write tests involving the file system.

emersonknapp, 2023-07-12:
Not sure if this is exactly what you need, but there is an event that is published whenever the Recorder splits to a new file.

https://github.com/ros2/rosbag2/blob/rolling/rosbag2_interfaces/msg/WriteSplitEvent.msg is published on  `~/events/write_split` with the path of the file just closed and the new file opened. The closed file will never be opened again by the recording process, so maybe subscribing to this topic meets your use case? 

Unfortunately I didn't find this feature documented anywhere which means you couldn't have known it existed!

ianstarz, 2023-07-13:
> subscribing to this topic meets your use case?
That very well may do the trick. We still want a process running outside of ROS to be able to pick up the change, but we could write a thin wrapper around this event that does that.

closed 2023-07-13 by ianstarz