ros2/rosbag2 #1429: Recording a bag file from a launch file.

DLu, 2023-07-20:
## Feature request

#### Feature description
The syntax for recording a bag file from a launch file always seemed awkward to me. 

https://github.com/ros2/examples/blob/1d97c4fc7445554f6f85f63305d424fc017212a0/launch_testing/launch_testing_examples/launch_testing_examples/record_rosbag_launch_test.py#L43-L46

Passing a raw ExecuteProcess command seems to invite errors. 

Instead, there should be a custom launch_ros action that does the same thing.

#### Implementation considerations
Seems like it could be a straightforward extension of ExecuteProcess.

MichaelOrlov, 2023-07-22:
@DLu Not sure what is expected behavior for the rosbag2 in this case. Could you please clarify or put some examples what would be expected behavior?

We might already be taking care of this enhancements suggestion in https://github.com/ros2/rosbag2/pull/1419 please take a look at it.

emersonknapp, 2023-07-25:
#1419 would allow for e.g. (syntax probably not 100% correct)

```
        launch_ros.actions.Node(
            executable='recorder',
            package='rosbag2',
            name='my_recorder',
            parameters=[{
                'all': True
            }]
        ),
```

Is that kind of what you were hoping for @DLu , or something different?

MichaelOrlov, 2023-08-22:
- Closing as a duplicate for #1419

closed 2023-08-22 by MichaelOrlov