ros2/rosbag2 #1433: when i try to write node to record message by homemade code,fail to load mcap plugin

Xen0sWang, 2023-07-30:
[ERROR] [rosbag2_storage]: Could not open '/home/udeer/Desktop/workspaces/bags/rosbag2_1_1_1/rosbag2_1_1_1_0' with 'mcap'. Error: bad file (get_interface_instance() at /home/udeer/Desktop/workspaces/ros2_ws/src/urecord/deps/src/rosbag2-0.12.0/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:79)
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'mcap'. (open_read_write() at /home/udeer/Desktop/workspaces/ros2_ws/src/urecord/deps/src/rosbag2-0.12.0/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:114)
terminate called after throwing an instance of 'std::runtime_error'
  what():  No storage could be initialized. Abort

Xen0sWang, 2023-07-30:
i try to rewrite Recorder class,and part of the code is as follows：
URecorder::URecorder(const std::string& node_name,
                     const rclcpp::NodeOptions& node_options)
    : rclcpp::Node(node_name, node_options),
      storage_options_(rosbag2_storage::StorageOptions{
          .storage_id = "mcap",
          .max_bagfile_size = 0,
          .max_bagfile_duration = 0,
          .max_cache_size = 100 * 1024 * 1024,
          .storage_preset_profile = "",
          .storage_config_uri = std::string{now().seconds()},
      }),
      record_options_(rosbag2_transport::RecordOptions{
          .all = false,
          .is_discovery_disabled = false,
          .topics = std::vector<std::string>{"/rosout"},
          .rmw_serialization_format =
              std::string(rmw_get_serialization_format()),
          .topic_polling_interval = 100ms,
          .regex = "",
          .exclude = "",
          .node_prefix = "_ros2cli",
          .compression_mode = "",
          .compression_format = "",
          .compression_queue_size = 1,
          .compression_threads = 0,
          .topic_qos_profile_overrides{},
          .include_hidden_topics = false,
      }),
      writer_(std::make_unique<UWriter>(
          std::make_unique<writers::USequentialWriter>())),
      slice_(create_wall_timer(150s, std::bind(&URecorder::slice, this))) {
  RCLCPP_INFO(get_logger(), "Init success");
  record();
}

emersonknapp, 2023-08-01:
Hi @Xen0sWang - I've reformatted your posts for code formatting (note: you use ``` triple backticks for this)

Also, I updated your description with the bug report template. Please fill out all of the information there so that we can understand how to help you. In the future, please use templates when provided, as it will enable more helpful responses.

One note: You are using `rosbag2_storage` version `0.12.0`, which isn't the active release for any past or present ROS 2 distribution. Perhaps this is from some old source checkout? As far as I know, 0.12.0 never went into an official distro (last Galactic version was 0.9 and first Humble version was 0.15), so maybe this is some very old source checkout or Rolling nightly package?

Xen0sWang, 2023-08-03:
Thanks for the amendment,I use the galactic version ros2,but i need some feature which rosbag2 0.9.0 don't have,so i update the rosbag2 version to 0.12.0

emersonknapp, 2023-08-03:
>Thanks for the amendment,I use the galactic version ros2,but i need some feature which rosbag2 0.9.0 don't have,so i update the rosbag2 version to 0.12.0

Please can you go fill _all_ of the fields in the issue description? I still don't quite understand - did you install Galactic from APT, then check out a later source version of rosbag2 and build just that? Are you running on Ubuntu Focal or Jammy? (etc)

MichaelOrlov, 2024-01-30:
- Closing as a stale issue with no response from the author.
- Can't reproduce.

closed 2024-01-30 by MichaelOrlov