ros2/rosbag2 #1435: Cannot access the data on rosbag2 using sqlite3

uljadberdica1000, 2023-08-07:
## Description
I cannot access the data in the topics

```
id                 name                                            type  \
0   1    [/control_pressure](https://file+.vscode-resource.vscode-cdn.net/control_pressure)          control_interfaces/msg/ControlPressure   
1   2     [/actual_pressure](https://file+.vscode-resource.vscode-cdn.net/actual_pressure)  pneucontrol_interfaces/msg/PressureSystemState   
2   3  [/mocap_rigid_bodies](https://file+.vscode-resource.vscode-cdn.net/mocap_rigid_bodies)   mocap_optitrack_interfaces/msg/RigidBodyArray   

  serialization_format                               offered_qos_profiles  
0                  cdr                                                     
1                  cdr  - history: 1\n  depth: 10\n  reliability: 1\n ...  
2                  cdr                                                     
```

I just need a way to deserialize and extract the data

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Install Method: apt
 - Version: [release, branch, commit hash, patch release number]

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2023-08-07:
Hi @uljadberdica1000 there is not enough information in the issue description about what you are trying to do and how to reproduce this issue.  Please provide more information.

From the issue description I see that issue was reproduced on the `Foxy` ROS 2 distro. Please be aware that `Foxy` distro has already reached EOL and we don't support it anymore. Please try to reproduce your problem on the newer ROS 2 distros such as Humble, Iron or Rolling.

Just in case, If you just don't know how to read messages from bag file - this is not a bug.  
Please refer to our unit tests 
https://github.com/ros2/rosbag2/blob/fda5aea264b4183d94352a7c7caf266ae104aa52/rosbag2_tests/test/rosbag2_tests/test_rosbag2_cpp_api.cpp#L103-L137
as examples of how to read and deserialize messages from bag files using `rosbag2_cpp` API.

MichaelOrlov, 2024-01-30:
- Closing as a stale issue with no response from the author.

closed 2024-01-30 by MichaelOrlov