ros2/rosbag2 #1437: :farmer: test_rosbag2_play_end_to_end flaky on Windows

Crola1702, 2023-08-08:
## Description

Flaky test [test_rosbag2_play_end_to_end](https://ci.ros2.org/view/nightly/job/nightly_win_deb/2813/testReport/junit/(root)/projectroot/test_rosbag2_play_end_to_end/) in windows CI (debug, release and repeated)

Test regressions:
* [projectroot.test_rosbag2_play_end_to_end](https://ci.ros2.org/view/nightly/job/nightly_win_deb/2813/testReport/junit/(root)/projectroot/test_rosbag2_play_end_to_end/)
* [rosbag2_tests.test_rosbag2_play_end_to_end.gtest.missing_result](https://ci.ros2.org/view/nightly/job/nightly_win_deb/2813/testReport/junit/(root)/rosbag2_tests/test_rosbag2_play_end_to_end_gtest_missing_result/)

### Expected Behavior
Test should pass

### Actual Behavior
Test failing because of a timeout

## To Reproduce
1. Run a build in [nightly_win_deb](https://ci.ros2.org/view/nightly/job/nightly_win_deb/), [nightly_win_rel](https://ci.ros2.org/view/nightly/job/nightly_win_rel/) or [nightly_win_rep](https://ci.ros2.org/view/nightly/job/nightly_win_rep/)
2. See projectroot.test_rosbag2_play_end_to_end fail

## System (please complete the following information)
 - OS: Windows
 - ROS 2 Distro: Rolling
 - Install Method: Source
 - Version: Rolling latest

## Additional context

Reference build: https://ci.ros2.org/view/nightly/job/nightly_win_deb/2813/

Test regressions:
* [projectroot.test_rosbag2_play_end_to_end](https://ci.ros2.org/view/nightly/job/nightly_win_deb/2813/testReport/junit/(root)/projectroot/test_rosbag2_play_end_to_end/)
* [rosbag2_tests.test_rosbag2_play_end_to_end.gtest.missing_result](https://ci.ros2.org/view/nightly/job/nightly_win_deb/2813/testReport/junit/(root)/rosbag2_tests/test_rosbag2_play_end_to_end_gtest_missing_result/)

Test is failing because of a timeout:

Log output:
```
[WARN] [1691491268.193064100] [subscription_manager_1691491266065392000]: SubscriptionManager::continue_spinning(..) finished by timeout
C:\ci\ws\src\ros2\rosbag2\rosbag2_tests\test\rosbag2_tests\test_rosbag2_play_end_to_end.cpp(80): error: Expected equality of these values:
  future.wait_for(service_call_timeout_)
    Which is: 4-byte object <01-00 00-00>
  std::future_status::ready
    Which is: 4-byte object <00-00 00-00>
 1/11 Test  #2: test_rosbag2_play_end_to_end .....***Timeout  60.00 sec
```

Test gets stuck when the error pops up (normally takes 15 seconds to run)

---

#### Flakiness ratio

job_name|last_time|first_time|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rep|2023-08-08|2021-07-26|15|9|60.0
nightly_win_rel|2023-08-08|2021-07-24|15|5|33.33
nightly_win_deb|2023-08-08|2021-07-25|16|5|31.25

Crola1702, 2023-08-17:
@clalancette, this is happening almost consistently in Windows builds. Do you think it's a good time to disable this test while https://github.com/ros2/rosbag2/pull/1342 is resolved? (it seems it would be a long time until we have updates on the fix PR)

MichaelOrlov, 2023-08-22:
@Crola1702 @clalancette I guess it hasn't been fully fixed for Windows after my overhaul in player tests per https://github.com/ros2/rosbag2/pull/1297.  
I will try to take a look at it this week. 
If I will not find a solution we can disable it for Windows again.

MichaelOrlov, 2023-08-22:
BTW. I have come across with this failure in one of my recent PR https://github.com/ros2/rosbag2/pull/1423#issuecomment-1666421538.  
My preliminary analysis was:
> Failed PlayEndToEndTestFixture.play_filters_by_topic by timeout since was not able to receive confirmation about service call for player resume in 60 seconds.

Crola1702, 2023-08-22:
> BTW. I have come across with this failure in one of my recent PR https://github.com/ros2/rosbag2/pull/1423#issuecomment-1666421538.

I haven't seen this failure in the buildfarm :thinking:

Crola1702, 2023-08-23:
I'm disabling the test in: https://github.com/ros2/rosbag2/pull/1452

MichaelOrlov, 2024-09-07:
Preliminary analysis:
There is a race condition between when service became available and when we start spining player node to be able to process service callbacks. i.e. it turns out that `resume` service created and on the other side we check in the test that service is available via graph and sending service request. However the rosbag2 player may not be ready to process callbacks from the services since we haven't yet even created executor for spinning player node.

The solution would be to move `play/pause` service creation from player constructor to the `roabg2_palyer->play` method.

cc: @clalancette

clalancette, 2024-09-12:
@MichaelOrlov Could this be another case similar to https://github.com/ros2/rosbag2/pull/1796#pullrequestreview-2299314606 ?

MichaelOrlov, 2024-09-12:
@clalancette Not the same. It is completely different. 
Here, the problem is that we first constructing a Rosbag2 Player class in the `rosbag2_py::transport_.cpp` in the constructor we creating service for `play/pause` and this service became visible via node_graph for the test wich is waiting for it in another process. The test start sending service request for resume playback since it see that service became available. Although on another side (Player) we haven't yet started `Player::Play()` and(or) haven't yet created executor to spin it in the `rosbag2_py::transport_.cpp`.
https://github.com/ros2/rosbag2/blob/2d4d02fd1374781f7111da3b0d91777905d9f7be/rosbag2_py/src/rosbag2_py/_transport.cpp#L262-L271

MichaelOrlov, 2024-09-12:
It seems I found a workaround for tests.
In test before sending `successful_service_request<Resume>(cli_resume_);` request 
https://github.com/ros2/rosbag2/blob/2d4d02fd1374781f7111da3b0d91777905d9f7be/rosbag2_tests/test/rosbag2_tests/test_rosbag2_play_end_to_end.cpp#L195-L197

Need repeatedely request for another service `IsPaused`, but ignoring the failure if we will not get response. When we will get response for `IsPaused` from Player - it means that Rosbag2 Player is fully started and ready. It is sort of current ststus of the Player. 
Will need to write some helper function similar to the 
https://github.com/ros2/rosbag2/blob/2d4d02fd1374781f7111da3b0d91777905d9f7be/rosbag2_tests/test/rosbag2_tests/test_rosbag2_play_end_to_end.cpp#L76-L89

MichaelOrlov, 2024-09-12:
@r7vme Help here will be appreciated.