ros2/rosbag2 #1444: Do not warn for unknown types if they are not included in the topic list

Aposhian, 2023-08-14:
## Description
If there is a topic with an unknown type on the ROS domain, `ros2 bag record` spams messages like
```
[WARN 1692051327.854323893] [ROSBAG2_TRANSPORT]: Topic '/performance_metrics' has unknown type 'gazebo_msgs/msg/PerformanceMetrics' . Only topics with known type are supported.
```
_every single time a message is received_, even if that topic wasn't even in the list of topics to record.

## Related Issues
It should also probably be a separate enhancement to only log this message once for a particular topic, since logging on every message is a lot of logs, and doesn't add any information.

## Completion Criteria
If a topic is not specified in the topic lists, do not warn about issues with it.

## Implementation Notes / Suggestions
Could just filter by topic list right before this message.

## Testing Notes / Suggestions

MichaelOrlov, 2023-09-12:
- Closing as resolved in the https://github.com/ros2/rosbag2/pull/1018
Now the warning about unknown topic types will appear only once per each topic during recording.

MichaelOrlov, 2023-09-12:
@Jannkar Thanks for the heads up. 
Indeed we do have a check if topics are not on the list but the checks are ordered such a way that we are checking for unknown types first.

I made a quick fix for that in https://github.com/ros2/rosbag2/pull/1466. 
However it is in the draft yet, I am trying to figure out if will be possible to create unit tests for regression.

PR #1466 (2023-09-13): Don't warn for unknown types if topics are not selected
- Closes #1444

Move the check for the unknown types to the bottom of the checks in the topic_filter
closed 2023-09-13 by MichaelOrlov