ros2/rosbag2 #1449: Add additional examples for reading ROSBag data and serialized messages

MaxFleur, 2023-08-22:
## Description

Currently, the ROSBag2 examples contain three different examples for writing data to a ROSBag via C++  or Python. However, there are no specific examples of reading data from ROSBags (neither for the standard reader or the sequential reader). In addition, the writing examples do not show how to serialize various messages before writing them to a bag file.
Inside the file under _rosbag2_tests/test/rosbag2_tests/test_rosbag2_cpp_api.cpp_, there are test cases for reading data and also for the message serialization. These helped me when using the ROSBag2 API, but in my opinion, a better idea is to create dedicated examples, because the unit tests should be meant solely for testing.

## Completion Criteria

- Example for reading ROSBag data using a simple reader (`rosbag2_cpp::Reader`)
- Example for reading ROSBag data using a sequential reader (`rosbag2_cpp::SequentialReader`)
- Example for writing ROSBag data using serialized messages (`rclcpp::SerializedMessage` and `rclcpp::Serialization<>`)

Ryanf55, 2024-02-16:
This one covers reading:
https://docs.ros.org/en/rolling/Tutorials/Advanced/Reading-From-A-Bag-File-CPP.html

Unfortunately, if the topic name is the wrong topic type than what is attempted to deserialize to, then there is a crash.

fujitatomoya, 2024-02-16:
IMO we could have `rosbag2_examples_<lang>::SimpleBagPlayer` and https://docs.ros.org/en/rolling/Tutorials/Advanced/Reading-From-A-Bag-File-CPP.html link it as an example.

This seems `good-first-issue`, probably we can tag this issue with `help-wanted`? 😄

sangteak601, 2024-02-28:
I want to take a look at this.
@MichaelOrlov Do you mind assigning this to me ?

MichaelOrlov, 2024-02-28:
@sangteak601 Thank you for being willing to help with this. You have got it :wink:

closed 2024-06-26 by MichaelOrlov