ros2/rosbag2 #1450: StorageOptions max_bagfile_duration is not working - creates a new bag file at a much faster rate than specified

cnicho35, 2023-08-23:
## Description
I am trying to make it so the node makes a new bag file every 10 mins but when I do this, it makes a new bag file more than once per second.
Here is my definition of the StorageOptions structure:
```
      std::string bag_path = filepath_param.as_string() + stampToString(tsOpen);
      uint64_t filelength_sec = 600;
      uint64_t zero = 0;

      rosbag2_storage::StorageOptions storage_options({bag_path, "sqlite3", zero , filelength_sec}); /** max duration in seconds = */

      writer.open(storage_options);
```

### Expected Behavior

It should make a new db3 file in the bag folder every 600 seconds

### Actual Behavior
It makes a new bag file in the folder more than once per second.

## System (please complete the following information)
 - OS: [e.g. Ubuntu Jammy]
 - ROS 2 Distro: Humble
 - Install Method:  APT
 - Version:  Version: 0.15.7-1jammy.20230807.182931

## Additional context

```
 md@ md-UP:~/M2-Dev/ md-ROS2-HERO$ ros2 launch  md_hero_launch start.launch.py 
[INFO] [launch]: All log files can be found below /home/ md/.ros/log/2023-08-23-08-24-50-947967- md-UP-7019
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [microstrain_inertial_driver_node-1]: process started with pid [7030]
[INFO] [ads1115_ros_node-2]: process started with pid [7032]
[INFO] [data_aggregator_node-3]: process started with pid [7034]
[microstrain_inertial_driver_node-1] [INFO] [1692800691.628009964] [microstrain_inertial_driver]: Running microstrain_inertial_driver version: unknown
[microstrain_inertial_driver_node-1] [INFO] [1692800691.630032357] [microstrain_inertial_driver]: Using MIP SDK version: v0.0.0
[microstrain_inertial_driver_node-1] [INFO] [1692800691.634507407] [microstrain_inertial_driver]: Attempting to open serial port </dev/ttyUSB1> at <230400>
[microstrain_inertial_driver_node-1] [INFO] [1692800691.636817339] [microstrain_inertial_driver]: Setting device to idle in order to configure
[microstrain_inertial_driver_node-1] [INFO] [1692800691.639246237] [microstrain_inertial_driver]: Note: Setting UART baudrate to 230400
[microstrain_inertial_driver_node-1] [INFO] [1692800691.928304147] [microstrain_inertial_driver]: Main Connection Info:
[microstrain_inertial_driver_node-1]     #######################
[microstrain_inertial_driver_node-1]     Model Name:       3DM-GX5-45
[microstrain_inertial_driver_node-1]     Serial Number:    6251.102720
[microstrain_inertial_driver_node-1]     Firmware Version: 1.1.69
[microstrain_inertial_driver_node-1]     #######################
[microstrain_inertial_driver_node-1] [INFO] [1692800691.930735361] [microstrain_inertial_driver]: Note: Not opening aux port because RTK dongle enable was not set to true.
[microstrain_inertial_driver_node-1] [INFO] [1692800691.931773534] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/aiding_status
[microstrain_inertial_driver_node-1] [INFO] [1692800691.931818196] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/antenna_offset_correction
[microstrain_inertial_driver_node-1] [INFO] [1692800691.931848823] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/rf_error_detection
[microstrain_inertial_driver_node-1] [INFO] [1692800691.931871086] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss1/sbas_info
[microstrain_inertial_driver_node-1] [INFO] [1692800691.931893882] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/aiding_status
[microstrain_inertial_driver_node-1] [INFO] [1692800691.931951971] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/antenna_offset_correction
[microstrain_inertial_driver_node-1] [INFO] [1692800691.931973643] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/fix
[microstrain_inertial_driver_node-1] [INFO] [1692800691.931994763] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/fix_info
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932015547] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/odom
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932036603] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/rf_error_detection
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932057433] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/sbas_info
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932078330] [microstrain_inertial_driver]: Note: The device does not support publishing the topic gnss2/time_ref
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932102795] [microstrain_inertial_driver]: Note: The device does not support publishing the topic imu/overrange_status
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932126854] [microstrain_inertial_driver]: Note: The device does not support publishing the topic nav/aiding_summary
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932148865] [microstrain_inertial_driver]: Note: The device does not support publishing the topic nav/dual_antenna_status
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932189933] [microstrain_inertial_driver]: Note: The device does not support publishing the topic rtk/status
[microstrain_inertial_driver_node-1] [INFO] [1692800691.932213886] [microstrain_inertial_driver]: Note: The device does not support publishing the topic rtk/status_v1
[microstrain_inertial_driver_node-1] [INFO] [1692800692.177681400] [microstrain_inertial_driver]: Setting spin rate to <100.000000> hz
[microstrain_inertial_driver_node-1] [INFO] [1692800692.180200258] [microstrain_inertial_driver]: Resuming the device data streams
[microstrain_inertial_driver_node-1] [INFO] [1692800692.183215661] [microstrain_inertial_driver]: Node activated
[data_aggregator_node-3] [INFO] [1692800692.427184225] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_0.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.478692922] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_1.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.547963692] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_2.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.569055280] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_3.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.647121255] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_4.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.669024568] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_5.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.747324211] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_6.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.768263358] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_7.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.844431210] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_8.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.868060546] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_9.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.945525092] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_10.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800692.967590462] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_11.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800693.046173806] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_12.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800693.067585510] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_13.db3' for READ_WRITE.
[... cut ...]
[data_aggregator_node-3] [INFO] [1692800693.245478326] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_16.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800693.267592413] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_17.db3' for READ_WRITE.
[data_aggregator_node-3] [INFO] [1692800693.347993260] [rosbag2_storage]: Opened database '/home/ md/Data/20230823_142452_MB.1692/20230823_142452_MB.1692_18.db3' for READ_WRITE.
```

cnicho35, 2023-08-29:
I think I figured out the issue - some of my timestamps in the header of my messages were way off making it think that the time was 1970.Are you using the timestamp in the header to determine how much time has elapsed since the bag was open?

MichaelOrlov, 2023-09-06:
@cnicho35 Bag split uses timestamps received from the node's clock at the time when the message arrived 
https://github.com/ros2/rosbag2/blob/a48001a33cf87bb812e9cc246e2a5da5283f0ed5/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L494-L503

https://github.com/ros2/rosbag2/blob/a48001a33cf87bb812e9cc246e2a5da5283f0ed5/rosbag2_cpp/src/rosbag2_cpp/writers/sequential_writer.cpp#L363-L365

MichaelOrlov, 2023-12-30:
- Closing as WAD (Works As Designed)

closed 2023-12-30 by MichaelOrlov