ros2/rosbag2 #1458: python node with Recorder does not exit with CTRL-C

tonynajjar, 2023-09-05:
## Description
A rclpy node instantiating a `rosbag2_py.Recorder` does not exit with CTRL-C

### Expected Behavior
It should exit cleanly

### Actual Behavior
It just hangs

## To Reproduce
```
import rclpy

from rclpy.node import Node
from rosbag2_py import Recorder

class RosBagRecorder(Node):
    def __init__(self):
        super().__init__("rosbag_recorder")
        self.recorder = Recorder()

def main():
    rclpy.init()
    node = RosBagRecorder()
    rclpy.spin(node)

if __name__ == "__main__":
    main()
```

## System (please complete the following information)
 - OS: Ubuntu 22
 - ROS 2 Distro: Iron
 - Install Method: source
 - Version: iron latest

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2023-09-07:
@tonynajjar The code behave as expected by design.  
It doesn't make sense to create rosbag2 recorder and spin node directly.  
We have SingleThreadedExecuter inside `Recorder::record(..)` method which will spin the node and gracefully handle `SIGINT` and `SIGTERM` signals. 

in your example, you are missing call for the 
```
recorder.record(storage_options, record_options, "node_name")
```

Please refer to the https://github.com/ros2/rosbag2/blob/dcd239ceea842752c344f0048572deb5f502feb8/ros2bag/ros2bag/verb/record.py#L255-L260
as an example of how to properly use it.

MichaelOrlov, 2023-09-07:
@tonynajjar Yes it is true. We override signal handlers in the recorder constructor to handle them further on in blocking `recorder::record(..)` method.   
Perhaps we can change this behavior and install signal handlers at the beginning of the `recorder::record(..)` method.

Out of curiosity what would be a use case for instantiating `Recorder()` instance and not using `recorder::record(..)` method after that?

tonynajjar, 2023-09-07:
Well the use case is that I want to make a Node which has a service server that when called will trigger the recording of a bag. For the recording I use this recorder, but it only starts recording when the service is called

tonynajjar, 2023-09-07:
The use case is I want to have a node with a service server that when called, will start recording a bag with the help of the Recorder. A bit like what is asked in https://github.com/ros2/rosbag2/issues/685

So the Recorder exists but we only start recording when the service is called

fujitatomoya, 2024-01-02:
> The use case is I want to have a node with a service server that when called, will start recording a bag with the help of the Recorder.

@tonynajjar https://github.com/ros2/rosbag2?tab=readme-ov-file#controlling-recordings-via-services does not work for your use case?

PR #1464 (2024-01-04): Install signal handlers in recorder only inside record method
- Closes #1458 

Notes for reviewers:
- It turns out that a possible situation when Recorder instance is instantiated and needs to spin the node for some time before calling `Recorder::record(..)` method for instance handling some service events before initiating recording. During those times signal handlers shall not be overridden.
- I removed rclcpp initialization with `rclcpp::SignalHandlerOptions::None` and `shutdown_on_signal` options since it doesn't have any effect because we are replacing signals with our own signal handlers [... cut]
closed 2024-01-04 by MichaelOrlov