ros2/rosbag2 #1462: Record ROS2 Parameters within a ros2 bag

vinnnyr, 2023-09-08:
## Description

I would love it if we could store ROS2 Parameters that were present during runtime into a bag so that is archived in . Often I have trouble looking at bags of, let's say controller performance, but not having record of what controller parameters we used to achieve such performance. 

## Related Issues

I think in order to handle parameters changed DURING the bag recording, we need to also do https://github.com/ros2/rosbag2/issues/773 but it's not strictly blocking

## Completion Criteria

-  All ROS2 parameters are saved at record time via optional CLI arg
- Thes ROS2 parameters can be queryied / inspected after recording is done

## Implementation Notes / Suggestions

** If you have ideas about how this feature might be accomplished, put them here. Note that this is just a suggestion to the implementer, so feel free to speculate. **
Maybe this is something to store in metadata.yaml? maybe it can take advantage of the key value pair in the bag?

## Testing Notes / Suggestions

** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

MichaelOrlov, 2023-09-09:
@vinnnyr The key problem here is how to get those parameters from the remote nodes.
Any ideas?

Pan-Navigator, 2026-04-20:
Hi,

I would like to discuss and if possible help out with the implementation!
Starting with a naive approach, how about a flag to record params on a latched topic within the rosbag