ros2/rosbag2 #1467: drop rate

laurafbec, 2023-09-13:
## Description
High messages' drop with high rate topics

### Expected Behavior
I've developed a node like [this](https://docs.ros.org/en/galactic/Tutorials/Advanced/Recording-A-Bag-From-Your-Own-Node-Py.html) to record messages belonging to several topics.

### Actual Behavior
I'm experiencing high messagess losses regarding topics with high rates as /tf, /odom or /scan. 
These are their QoS definitions:

` self.scan_qos = QoSProfile(
          durability=QoSDurabilityPolicy.VOLATILE,
          reliability=QoSReliabilityPolicy.BEST_EFFORT,
          history=QoSHistoryPolicy.KEEP_LAST,
          depth=500)

 self.odom_qos = QoSProfile(
          durability=QoSDurabilityPolicy.VOLATILE,
          reliability=QoSReliabilityPolicy.RELIABLE,
          history=QoSHistoryPolicy.KEEP_LAST,
          depth=100)

self.tf_qos = QoSProfile(
          durability=QoSDurabilityPolicy.VOLATILE,
          reliability=QoSReliabilityPolicy.RELIABLE,
          history=QoSHistoryPolicy.KEEP_LAST,
          depth=200)
`

## System (please complete the following information)
 - OS: Ubuntu
 - ROS 2 Distro: Humble
 - Install Method: APT

MichaelOrlov, 2023-09-14:
@laurafbec Using storage_id='mcap' might help with performance. 
You can find some comparisons at the https://www.apex.ai/post/improvements-in-data-recording-path in "Test results for serialized data over shared memory" chapter

laurafbec, 2023-09-15:
Thanks @MichaelOrlov!!

MichaelOrlov, 2024-01-30:
@laurafbec As regards
> How do you estimated the sent messages? By multiplying rate by time?

No. I was using [rosbag2_performance_benchmarking](https://github.com/ros2/rosbag2/tree/rolling/rosbag2_performance/rosbag2_performance_benchmarking) package and was known exactly how many messages were sent and how many received per each iteration.

- Closing this issue as a stale issue. 
- Performance has been improved by introducing the new `MCAP` storage format.

closed 2024-01-30 by MichaelOrlov