ros2/rosbag2 #1470: `Failed to deserialize ROS message` is thrown if bag file with a JointState message is read

MaxFleur, 2023-09-14:
## Description
For a project, I am writing messages of the types `std_msgs/msg/int32` and `sensor_msgs/msg/JointState` to a ROSBag. The bag file contains both respective topics and if these topics are listened to via `ros2 topic echo`, both topics will show valid data.
In a later part of the program, I want to read the data from the bag file. In the case of the integer data, this works fine. However, if I want to read the JointState data, an error message is thrown:
`Failed to deserialize ROS message.: Fast CDR exception deserializing message of type sensor_msgs::msg::dds_::JointState_., at ./src/type_support_common.cpp:118`.

I have written a minimal example with this bug. The code is as follows:
```
#include <iostream>

#include "rosbag2_cpp/reader.hpp"

#include "std_msgs/msg/int32.hpp"
#include "sensor_msgs/msg/joint_state.hpp"

int main(int argc, char * argv[])
{
    rosbag2_cpp::Reader reader;

    reader.open("/path/to/recording");
    while (reader.has_next()) {
        // This will work, constantly put out 0
        auto bagMessageInteger = reader.read_next<std_msgs::msg::Int32>();
        std::cout << bagMessageInteger.data << std::endl;

        // If this code is used instead, an error message will be thrown
        auto bagMessageJointState = reader.read_next<sensor_msgs::msg::JointState>();
        std::cout << bagMessageJointState.position[0] << std::endl;
    }
    return 0;
}
```

For better traceability, I've attached the code along with a packages.xml and CMakeLists.txt as well as an example ROSBag:
[example.tar.gz](https://github.com/ros2/rosbag2/files/12607278/example.tar.gz)

### Expected Behavior
The joint state data should be readable.

### Actual Behavior
An error message is thrown when trying to call `reader.read_next<sensor_msgs::msg::JointState>();`.

## To Reproduce
1. Download the attached file and extract it into your ROS2 workspace
2. Adjust the path to the ros2bag directory in line 12 of the main.cpp
3. Build and run the tool with `colcon build` and `ros2 run rosbag2_reading_test rosbag2_reading_test` -> Everything will work fine
4. Comment line 15 and 16 out and uncomment line 19 and 20
5. Repeat step 3 -> an error message will be thrown.

## System (please complete the following information)
 - OS: Ubuntu 22.04 LTS
 - ROS 2 Distro: Humble
 - Install Method: APT
 - Version: 0.15.7

MaxFleur, 2023-09-15:
One more interesting note: If no code is commented out (which means that both messages are extracted in one loop iteration), the program will run just fine. No idea why, though.

MichaelOrlov, 2023-09-20:
Hi @MaxFleur, This is not a bug  this is a feature.
The templated version of the `read_next<MessageT>()` API is not meant to be used as you want it to be used.
It could be safely used when bag contain only one message type.
See what inside of it
https://github.com/ros2/rosbag2/blob/64e29724cbea3058bfa4c1688867dc9f2f85bdbd/rosbag2_cpp/include/rosbag2_cpp/reader.hpp#L144-L154

- Closing this issue as wouldn't be fixed.

closed 2023-09-20 by MichaelOrlov

MaxFleur, 2023-09-20:
@MichaelOrlov 
Ok thanks for the info. It might be nice to better document this, though. The examples show only code for writing to a ROSBag, but not reading from it (only the tests do that currently, but they should not be meant to be read for understanding the basic code imo). 
I found one more example on the ROS2 Tutorials page for reading bag data from a node in C++. Might have overseen this a few days ago... but with that tutorial, it works now. 
Thanks for pointing it out anyway! :-)