ros2/rosbag2 #1477: ros2 bag record cmd - Invalid choice mcap in --storage flag

sagocz, 2023-10-03:
## Description
I want to use `.mcap` format in ros2 bag record. I'm using the newest `humble` version.

### Expected Behavior
`ros2 bag record` cmd with `.mcap` format writing.

### Actual Behavior
```
ros2 bag record: error: argument -s/--storage: invalid choice: 'mcap' (choose from 'my_test_plugin', 'sqlite3')
```
## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Humble

sagocz, 2023-10-04:
@MichaelOrlov 
It's ok. But I also tried to pull the newest docker image from osrf. -> [humble image](https://hub.docker.com/layers/osrf/ros/humble-desktop-full-jammy/images/sha256-b6ed48784c01667cbee94037c48f73bf603fe5a06b81fc29a4b11e140cc7179b?context=explore)
And when I run it, I sourced to `. /opt/ros/humble/setup.bash` and next type cmd `ros2 bag record -h`.  
I think there may be missing installation of specific packages from lack of dependencies in the sources. Going to the `/opt/ros/humble/share` directory, packages regarding `.mcap` are missing. On the other hand, downloading the iron version everything works as expected and you can record bags directly from `ros2 bag record`. What I find strange about this behavior is that when I go to [branch humble](https://github.com/ros2/rosbag2/tree/humble) in the ros2/rosbag2 repo, I see that in version 0.15.8 there are packages that support .mcap recording, while they cannot be used from the cmd command `ros2 bag record -s mcap`.
```
root@1f08929c461d:/# ros2 pkg list
action_msgs
action_tutorials_cpp
action_tutorials_interfaces
action_tutorials_py
actionlib_msgs
actuator_msgs
ament_cmake
ament_cmake_auto
ament_cmake_copyright
ament_cmake_core
ament_cmake_cppcheck
ament_cmake_cpplint
ament_cmake_export_definitions
ament_cmake_export_dependencies
ament_cmake_export_include_directories
ament_cmake_export_interfaces
ament_cmake_export_libraries
ament_cmake_export_link_flags
ament_cmake_export_targets
ament_cmake_flake8
ament_cmake_gen_version_h
ament_cmake_gmock
ament_cmake_gtest
ament_cmake_include_directories
ament_cmake_libraries
ament_cmake_lint_cmake
ament_cmake_pep257
ament_cmake_pytest
ament_cmake_python
ament_cmake_ros
ament_cmake_target_dependencies
ament_cmake_test
ament_cmake_uncrustify
ament_cmake_version
ament_cmake_xmllint
ament_copyright
ament_cppcheck
ament_cpplint
ament_flake8
ament_index_cpp
ament_index_python
ament_lint
ament_lint_auto
ament_lint_cmake
ament_lint_common
ament_package
ament_pep257
ament_uncrustify
ament_xmllint
angles
builtin_interfaces
camera_calibration
camera_calibration_parsers
camera_info_manager
class_loader
common_interfaces
composition
composition_interfaces
compressed_depth_image_transport
compressed_image_transport
console_bridge_vendor
cv_bridge
demo_nodes_cpp
demo_nodes_cpp_native
[... 220 lines cut ...]
tf2_ros
tf2_ros_py
tf2_sensor_msgs
tf2_tools
theora_image_transport
tinyxml2_vendor
tinyxml_vendor
tlsf
tlsf_cpp
topic_monitor
tracetools
tracetools_image_pipeline
trajectory_msgs
turtlesim
uncrustify_vendor
unique_identifier_msgs
urdf
urdf_parser_plugin
vision_opencv
visualization_msgs
xacro
yaml_cpp_vendor
zstd_vendor

```

clalancette, 2023-10-04:
> I think there may be missing installation of specific packages from lack of dependencies in the sources. Going to the `/opt/ros/humble/share` directory, packages regarding `.mcap` are missing. On [...]

This is more-or-less working as designed.  In Humble, the default backend for rosbag2 is sqlite3, which *is* installed above (in Humble, it is part of `rosbag2_storage_default_plugins`).  `rosbag2_storage_mcap` is not part of the default install.  At the end of the day, the variants (`core`, `base`, `desktop`, `desktop-full`, etc) are a tradeoff between keeping the installation small, and providing functionality.

With all of that said, I see 3 ways forward here:
1.  Just accept that mcap is not part of the default install in Humble, and leave it at that.  This may require a documentation update specifically for Humble.
2.  Add in `rosbag2_storage_mcap` as a dependency of one of the packages in https://github.com/ros2/rosbag2/tree/humble .  That will pull it in for all variants.  We'll want to carefully consider what the size impact on each of the variants is before we do this.
3.  Add in `rosbag2_storage_mcap` as a variant dependency in https://github.com/ros2/variants .  This has the benefit of being more targeted (we can make it a dependency only for certain variants), but it means that humble is "different" from all of the rest.

Personally, I think we should just stick with 1, with a possible documentation update.  MCAP is not the default in Humble.

clalancette, 2023-10-04:
> It's a pity you didn't add it for humble vs iron already.

It is added in both of them.  It is just not the default for humble.  You can always install it with `apt-get install ros-humble-rosbag2-storage-mcap`.

> I think that in the next version of ros the mcap format will already be the default or at least an optional form of recording.

Correct.  In both Iron and Rolling, `mcap` is the default.

> At this moment, as I understand using the humble version, I have no possibility in the near future (next month) to record in this format?

No, it works totally fine in Humble.  You just need to install it as I mentioned above.

closed 2023-10-06 by sagocz