ros2/rosbag2 #1484: When I perform the ros2 bag play operation, it will cause oom.

ablalala, 2023-10-18:
## Description
** clear and concise description of the bug is **
when i ros2 bag play the whole bag(like this:ros2 bag play bagname)，an i use top command to show  the RES, the largest res occupancy is ros2 which occupies 300M，when i ros2 bag play one topic from the bag(like this:ros2 bag play bagname --topic lidar/LIDAR_FL_CLOUD),an i use top command to show  the RES, the largest res occupancy is ros2 which occupies 1400M,This behavior will causes oom in the development board，in addition,the RES for playing other topic is also 300M.I would like to know why the memory footprint when playing the entire package is still smaller than playing a topic in the code packet of the released version
### Expected Behavior
** clear and concise description of what you _expected_ to happen **

### Actual Behavior
** clear and concise description of what _actually_ happened. include copied logs whenever possible **

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. Logs print error '...' **

## System (please complete the following information)
 - OS: [e.g. Ubuntu Bionic]
 - ROS 2 Distro: [e.g. Dashing]
 - Install Method: [APT, release archive, source]
 - Version: [release, branch, commit hash, patch release number]

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2023-10-22:
Hi @ablalala.
The reason why you see excessive memory consumption during playback with point cloud topics is that we need to have some caching read-ahead queue in the rosbag2 player for smooth playback. Since file read out usually a slow operation and blocking system call. 
However, we have a `--read-ahead-queue-size` CLI parameter for the player
https://github.com/ros2/rosbag2/blob/8216ea22daa78abd3fe8a1607a6fb91351e340e1/ros2bag/ros2bag/verb/play.py#L43-L47
As you can see the default value is **1000**. And if you have mostly point cloud messages in your recorded bag you will see memory allocation around **1000*sizeof(YourPointCloudMessage)**.

Another source of potential excessive memory consumption is the DDS internal queue depth(size) which is by default **10** for each topic and settling up as QoS parameter during pub/sub creations. 
Although we have the ability to override QoS settings for each topic during rosbag2 playback.
Please refer to the [overriding-qos-profiles](https://github.com/ros2/rosbag2#overriding-qos-profiles) section in the rosbag2 readme documentation for details on how to do this.

MichaelOrlov, 2023-12-30:
- Closing as a stale issue with an answer and no response from the author.

closed 2023-12-30 by MichaelOrlov