ros2/rosbag2 #1485: CPU overhead when discovery is used

doisyg, 2023-10-20:
## Description
Related to https://github.com/ros2/rosbag2/issues/1170
When recording a list of topics that are not all published, on a system with multiple topics, actions and services (typically a mobile robot with the nav2 stack), and with (default) automatic discovery, there is an important CPU usage from the bag recorder process.Typically 20% CPU usage on a very powerful machine (i7 13th gen)
We don't observe this CPU overhead with the same setup but `--no-discovery` added.

### Expected Behavior
Reasonable CPU usage

### Actual Behavior
Non reasonable CPU usage

## System (please complete the following information)
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Iron latest bin 
 - Install Method: apt
 - Version: 0.22.3-1jammy.20230911.053809

doisyg, 2023-10-20:
Update: no issue when using rosbag built from source and latest Iron commit: https://github.com/ros2/rosbag2/commit/7adc0f9da7bccb25d7eef56be69b393a9c3220d9
However when using source v0.22.3, the issue is there. So it feels that (good timing), the latest commit fix this issue.

Can you trigger a release ?

fujitatomoya, 2023-10-21:
@doisyg i am not sure which commit actually fixes your problem here. but for releasing new packages would take some time.

CC: @Yadunund

doisyg, 2023-10-21:
> i am not sure which commit actually fixes your problem here. but for releasing new packages would take some time.

I am not sure myself but I know that I have the problem with v0.22.3 (even from source), and I don't with the latest version of the Iron branch (so probably these changes: https://github.com/ros2/rosbag2/commit/51a83f4421ff335c2237e6214e9bb7e6ae206a92)

What's blocking for releasing the latest Iron branch in the next sync ?

MichaelOrlov, 2023-10-21:
@doisyg Yes, https://github.com/ros2/rosbag2/commit/51a83f4421ff335c2237e6214e9bb7e6ae206a92 might have improved the performance burden during discovery since we are differing more heavy check to the end of the list when filtering out topics.

We will spawn a new release for iron, however as others mentioned it could take some time.
Meanwhile to workaround the issue you can explicitly specify a list of topics for the rosabg2 recorder. In this case, as soon as all topics are subscribed we are exiting from the rosbag2 discovery thread and there are will not extra burden on the performance.

MichaelOrlov, 2024-05-05:
- Closing as fixed in the https://github.com/ros2/rosbag2/commit/51a83f4421ff335c2237e6214e9bb7e6ae206a92

closed 2024-05-05 by MichaelOrlov