ros2/rosbag2 #1492: [rosbag2_storage_mcap] Bugs in generated CMake targets?

EricCousineau-TRI, 2023-11-13:
## Description
Unable to depend on `rosbag2_storage_mcap` via CMake / ament.

### Expected Behavior
It should work.

### Actual Behavior
We get the following errors:
```
CMake Error at CMakeLists.txt:19 (add_library):
  Target "empty_using_rosbag2_storage_mcap" links to target
  "mcap_vendor::mcap" but the target was not found.  Perhaps a find_package()
  call is missing for an IMPORTED target, or an ALIAS target is missing?

CMake Error in CMakeLists.txt:
  Imported target "rosbag2_storage_mcap::rosbag2_storage_mcap" includes
  non-existent path

    "/opt/ros/humble/include/rosbag2_storage_mcap"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

## To Reproduce
Install ROS 2 Humble on Ubuntu 22.04.
Then run the following:
```
sudo apt install ros-humble-rosbag2-storage-mcap

mkdir /tmp/example && cd /tmp/example
cat <<'EOF'
cmake_minimum_required(VERSION 3.5)
project(empty_using_rosbag2_storage_mcap C CXX)

find_package(ament_cmake REQUIRED)
find_package(rosbag2_storage_mcap REQUIRED)

file(WRITE empty.cc "")
add_library(${PROJECT_NAME} SHARED empty.cc)

ament_target_dependencies(${PROJECT_NAME} rosbag2_storage_mcap)
target_link_libraries(${PROJECT_NAME} rosbag2_storage_mcap)
EOF

mkdir build && cd build
source /opt/ros/humble/setup.bash
cmake ..
```

## System (please complete the following information)
 - OS: Ubuntu Jammy 22.04
 - ROS 2 Distro: Humble
 - Install Method: Apt
 - Version: `0.15.8-1jammy.20230919.224231`

## Additional context
Root cause of https://github.com/RobotLocomotion/drake-ros/issues/315
Workaround in https://github.com/RobotLocomotion/drake-ros/pull/316

MichaelOrlov, 2023-11-13:
@EricCousineau-TRI I've briefly checked CMake.txt file for the `rosbag2_storage_macp` for reference to the `mcap_vendor` package
 and we do have both `find_package(mcap_vendor REQUIRED)` and `ament_target_dependencies(${PROJECT_NAME mcap_vendor)`
https://github.com/ros2/rosbag2/blob/0b4deb3311ed8397410d84b462a86b2d1c9db153/rosbag2_storage_mcap/CMakeLists.txt#L23-L42

I don't remember that we promised that any of ROS2 core packages would compile with the plain `cmake ..` command.
The proper way to compile code in ROS2 is using `colocn build` command.
Have you tried to reproduce the problem if you would create a proper ROS2 package that is dependent on `rosbag2_storage_mcap` and build it with `colcon build` ?

clalancette, 2023-11-13:
> I don't remember that we promised that any of ROS2 core packages would compile with the plain cmake .. command.

Actually, this has to work; that is how the buildfarm produces binary packages (it does not use `colcon`).

So it is curious that doing it "by hand" this way doesn't work.  It probably needs a bit more debugging to see what is going on here.

EricCousineau-TRI, 2023-11-13:
@MichaelOrlov The CMakeLists.txt you pointed out I believe is relevant for source builds, but not for consuming binaries installs of `rosbag2_storage_mcap` (using the installed artifacts).

I am able to reproduce this in a minimal colcon setup, which effectively shows the same behavior:
```sh
rm -rf /tmp/colcon-example && mkdir /tmp/colcon-example && cd /tmp/colcon-example
mkdir -p src/example

cat > src/example/CMakeLists.txt <<'EOF'
cmake_minimum_required(VERSION 3.14)
project(example)

find_package(ament_cmake REQUIRED)
find_package(rosbag2_storage_mcap REQUIRED)

file(WRITE empty.cc "")
add_library(${PROJECT_NAME} SHARED empty.cc)

ament_target_dependencies(${PROJECT_NAME} rosbag2_storage_mcap)
target_link_libraries(${PROJECT_NAME} rosbag2_storage_mcap)

ament_package()
EOF

cat > src/example/package.xml <<'EOF'
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>example</name>
  <version>0.0.1</version>
  <description>thing</description>
  <maintainer email="meh@nowhere.com">Meh</maintainer>
  <license>Apache-2.0</license>
  <buildtool_depend>ament_cmake</buildtool_depend>
  <depend>rosbag2_storage_map</depend>
  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
EOF

source /opt/ros/humble/setup.bash
colcon build
```

MichaelOrlov, 2023-11-13:
@EricCousineau-TRI Thanks for trying to reproduce with colcon build.
Now it seems I know why it fails to build.
We have missing dependencies in the `ament_export_dependencies(rosbag2_storage rcutils)`
https://github.com/ros2/rosbag2/blob/0b4deb3311ed8397410d84b462a86b2d1c9db153/rosbag2_storage_mcap/CMakeLists.txt#L111-L115
Perhaps the fix would be to add missing `mcap_vendor` and  `pluginlib` to the `ament_export_dependencies`.

The reason why it works in CI and other workflow is that we are using `rosbag2_storage_mcap` as a runtime loading library and never compiled directly against it.

ahcorde, 2023-11-14:
I openned a PR https://github.com/ros2/rosbag2/pull/1496

ahcorde, 2023-11-20:
The PR was merged and backported to `iron` and `humble`. Feel free to reopen the issue if the problem persists

closed 2023-11-20 by ahcorde