ros2/rosbag2 #1500: play service call doesn't return until play finished

KenYN, 2023-11-21:
## Description
When using the service interface to pull request https://github.com/ros2/rosbag2/pull/1419 (I'm not sure how to set up the same situation for `ros2 bag play`) with the command:

    $ rosservice call /bag_player/play rosbag2_interfaces/srv/Play "{start_offset: {sec: 0, nanosec: 0}, playback_duration: {sec: 0, nanosec: 0}, playback_until_timestamp: {sec: 0, nanosec: 0}}" requester: making request: rosbag2_interfaces.srv.Play_Request(start_offset=builtin_interfaces.msg.Time(sec=0, nanosec=0), playback_duration=builtin_interfaces.msg.Duration(sec=0, nanosec=0), playback_until_timestamp=builtin_interfaces.msg.Time(sec=0, nanosec=0))

### Expected Behavior
According to [this page](https://docs.ros.org/en/iron/How-To-Guides/Topics-Services-Actions.html#services) I would expect it to return quickly after starting the playback:

## Services

* Should be used for remote procedure calls that terminate quickly, e.g. for querying the state of a node or doing a quick calculation such as IK. They should never be used for longer running processes

### Actual Behavior
The service call does not return until play completes, and if looping is on, then it never returns until CTRL+C.

## System (please complete the following information)
Ubuntu 22.04 + Rolling

roncapat, 2023-11-21:
Very stupid question: does this happens only with my branch? I'm looking at the service callback code:

https://github.com/ros2/rosbag2/blob/7fcb7034fffb70a65b47ef222e432c53793491fb/rosbag2_transport/src/rosbag2_transport/player.cpp#L1080-L1092

and it seems to me also upstream rosbag2 waits for the end of `.play()` method.

MichaelOrlov, 2023-11-21:
@KenYN @roncapat Yeah, the `Player::play()` method is a blocking call and probably is not suitable for the direct call from the service callback.  
We may need to consider spawning a new thread for playback from service callback if we are not in the playback already.

PR #1503 (2023-11-29): Make `rosbag2_transport::Player::play()` run in a separate thread
- Closes #1500 

This is intended to solve the issue reported in #1500.
CC: @MichaelOrlov @KenYN

This can be also beneficial introductory work for #1419. As it has been noticed, playback shall start automatically when the `Player` component is loaded, but this can't be done via syncronous call by the component constructor.

Current status:
- separate thread implemented
- all tests preliminarly adapted

Anyway, I believe that a round of refactor can be done here. For example, direclty returning an `std::future` from `wait_for_pl [... cut]
closed 2023-11-29 by MichaelOrlov